dusty-nv/jetson-utils #6: Can this be applied to the Nano?

liukekoukele, 2019-07-03:
hello
Is this library compatible with nano correctly?  When I use nano, I cannot open camera correctly through C++ and opencv，  
camera module:Raspberry PI camera V2

vitcozzolino, 2019-08-06:
I use the library without issues on both TX2 and Nano but haven't tried using a camera.

dusty-nv, 2019-08-06:
Yes, works on Nano too.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

red24dog, 2019-08-09:
How is this package installed on a Nano? I note that it appears to be required to compile jetson-ross and deep-learning. What Dependancies must be installed? There is no "setup" or "configure".
Thanks