dusty-nv/jetson-utils #17: timeAdd() is incorrect resulting in Event::Wait() with timeout never timing out

jarnovanderlinden, 2019-12-06:
The timeAdd() function in timespec incorrectly calculates the carry from nsec to sec.
```
const time_t sec=t.tv_nsec/1e-9;
t.tv_sec+=sec;
t.tv_nsec-=sec*1e-9;
```
should probably be:
```
const time_t sec=t.tv_nsec/1e9;
t.tv_sec+=sec;
t.tv_nsec-=sec*1e9;
```
As this is used with the timeout versions of Event::Wait(), which in turn is used is gstCamera::Capture(), this bug makes the capture practically never time out.

dusty-nv, 2019-12-06:
Hi @jarnovanderlinden, sorry about that - I had previously patched this in [commit `6c8b6b`](https://github.com/dusty-nv/jetson-utils/commit/6c8b6b1b499dbb308095dead8c4b592560ad1228#diff-e4284d85734ecabbef00c399c442892a) to `jetson-utils`.  So if you pull the latest `jetson-utils` into your repo and recompile, it should be fixed.

``` bash
$ cd jetson-inference/utils
$ git pull origin master
$ cd ../build
$ cmake ../
$ make
$ sudo make install
```

Let me know if you still have an issue with it.

jarnovanderlinden, 2019-12-07:
Ah thank you. That does indeed fix it.
I missed the patches as I got here from jetson-inference, which uses an older revision.