dusty-nv/jetson-utils #23: I don´t want to see the print fron cudaFromNumpy()

Embring, 2020-04-12:
Hi!
Do anyone know if you can turnoff the print from cudaFromNumpy()?

tttdevelop, 2020-05-19:
Same problem ... I did the following and just recompile. (it's same procedure in the Jetson Inference code too)

Go to the file python/bindings/PyNumPy.cpp and delete or comment out the line 119;

`//printf(LOG_PY_UTILS` "cudaFromNumpy()  ndarray dim %i = %li\n", n, dims[n]);`

This disables the cudafromNumpy messages

and ...

Go to the file python/bindings/PyCUDA.cpp and delete or comment out the line 33;

`//printf(LOG_PY_UTILS "freeing CUDA mapped memory\n");`

and ...

Go to the file python/bindings/PyCUDA.cpp and delete or comment out the line 111;

`//printf(LOG_PY_UTILS` "freeing cudaMalloc memory\n");

Embring, 2020-07-11:
@tttdevelop  Thanks! :-)

dusty-nv, 2020-07-11:
If you check out the dev branch of jetson-inference, these are nonow disabled by default. And you can control the logging level from the command line.

Embring, 2020-07-22:
Thanks Dusty-nv!