dusty-nv/jetson-utils #27: RTSP support in videoSource does not currently support authenticated URIs.

drakorg, 2020-07-05:
Hi, I was trying the jetson-inference/utils native support for RTSP streams in the dev branch just to realize that it can't yet handle authenticated URIs, such as "rtsp://user:password@192.168.0.1" ...
Just to let you know of a kinda important feature.
Thank you.
Best regards.
Eduardo

dusty-nv, 2020-07-05:
Thanks Eduardo, do you know an example RTSP pipeline in gstreamer that works with authenticated RTSP?

I am unsure of how to pass the login credentials to the gstreamer RTSP or UDP elements, and unfortunately dont have an authenticated RTSP source to test it with.

drakorg, 2020-07-05:
Hi, TBH I have/had no idea how it worked under the hood, but a quick check
with wireshark showed this dialog between VLC and an IP cam I have here at
home:

REQUEST #1.
Frame 2262: 173 bytes on wire (1384 bits), 173 bytes captured (1384 bits)
on interface \Device\NPF_{51DF3F58-2ACC-4E77-B661-7F5A3C1BC285}, id 0
Ethernet II, Src: IntelCor_0e:d3:eb (60:6c:66:0e:d3:eb), Dst:
LSDScien_ae:51:9f (00:95:69:ae:51:9f)
Internet Protocol Version 4, Src: 192.168.0.62, Dst: 192.168.0.25
Transmission Control Protocol, Src Port: 9156, Dst Port: 554, Seq: 1, Ack:
1, Len: 119
Real Time Streaming Protocol
    Request: OPTIONS rtsp://192.168.0.25:554/ RTSP/1.0\r\n
    CSeq: 2\r\n
    User-Agent: LibVLC/3.0.10 (LIVE555 Streaming Media v2016.11.28)\r\n
    \r\n

RESPONSE #1:
Frame 2264: 207 bytes on wire (1656 bits), 207 bytes captured (1656 bits)
on interface \Device\NPF_{51DF3F58-2ACC-4E77-B661-7F5A3C1BC285}, id 0
Ethernet II, Src: LSDScien_ae:51:9f (00:95:69:ae:51:9f), Dst:
IntelCor_0e:d3:eb (60:6c:66:0e:d3:eb)
Internet Protocol Version 4, Src: 192.168.0.25, Dst: 192.168.0.62
Transmission Control Protocol, Src Port: 554, Dst Port: 9156, Seq: 1, Ack:
120, Len: 153
Real Time Streaming Protocol
    Response: RTSP/1.0 200 OK\r\n
    CSeq: 2\r\n
    Public: OPTIONS, DESCRIBE, PLAY, PAUSE, SETUP, TEARDOWN, SET_PARAMETER,
GET_PARAMETER\r\n
    Date:  Sun, Jul 05 2020 01:57:38 GMT\r\n
    \r\n

REQUEST #2:
Frame 2265: 199 bytes on wire (1592 bits), 199 bytes captured (1592 bits)
on interface \Device\NPF_{51DF3F58-2ACC-4E77-B661-7F5A3C1BC285}, id 0
Ethernet II, Src: IntelCor_0e:d3:eb (60:6c:66:0e:d3:eb), Dst:
LSDScien_ae:51:9f (00:95:69:ae:51:9f)
Internet Protocol Version 4, Src: 192.168.0.62, Dst: 192.168.0.25
Transmission Control Protocol, Src Port: 9156, Dst Port: 554, Seq: 120,
Ack: 154, Len: 145
Real Time Streaming Protocol
    Request: DESCRIBE rtsp://192.168.0.25:554/ RTSP/1.0\r\n
    CSeq: 3\r\n
    User-Agent: LibVLC/3.0.10 (LIVE555 Streaming Media v2016.11.28)\r\n
    Accept: application/sdp\r\n
    \r\n

RESPONSE #2:
Frame 2266: 288 bytes on wire (2304 bits), 288 bytes captured (2304 bits)
on interface \Device\NPF_{51DF3F58-2ACC-4E77-B661-7F5A3C1BC285}, id 0
Ethernet II, Src: LSDScien_ae:51:9f (00:95:69:ae:51:9f), Dst:
IntelCor_0e:d3:eb (60:6c:66:0e:d3:eb)
Internet Protocol Version 4, Src: 192.168.0.25, Dst: 192.168.0.62
Transmission Control Protocol, Src Port: 554, Dst Port: 9156, Seq: 154,
Ack: 265, Len: 234
Real Time Streaming Protocol
    Response: RTSP/1.0 401 Unauthorized\r\n
    CSeq: 3\r\n
    WWW-Authenticate: Digest realm="IP Camera(67156)",
nonce="d8b71bc1d5de46b7a797eb2e876f92e1", stale="FALSE"\r\n
    WWW-Authenticate: Basic realm="IP Camera(67156)"\r\n
    Date:  Sun, Jul 05 2020 01:57:38 GMT\r\n
    \r\n

REQUEST #3:
Frame 2267: 384 bytes on wire (3072 bits), 384 bytes captured (3072 bits)
on interface \Device\NPF_{51DF3F58-2ACC-4E77-B661-7F5A3C1BC285}, id 0
Ethernet II, Src: IntelCor_0e:d3:eb (60:6c:66:0e:d3:eb), Dst:
[... 90 lines cut ...]
Internet Protocol Version 4, Src: 192.168.0.25, Dst: 192.168.0.62
Transmission Control Protocol, Src Port: 554, Dst Port: 9156, Seq: 1427,
Ack: 1342, Len: 204
Real Time Streaming Protocol
    Response: RTSP/1.0 200 OK\r\n
    CSeq: 6\r\n
    Session: 1001209245;timeout=60
    Transport:
RTP/AVP;unicast;client_port=58442-58443;server_port=8226-8227;ssrc=50f1abfd;mode="play"
    Date:  Sun, Jul 05 2020 01:57:38 GMT\r\n
    \r\n

Apparently the server sends a nonce, and the client computes a response
based on the password ... and then sends this response in every subsequent
request ...
That's on the basics on the authentication mechanism ... regarding your
question about a pipeline that would make use of such authentication
mechanism I really don't know, I've just got in contact with gstreamer for
this project too, and so far I've been trying to avoid it but apparently
it'd appear to be the best option. If I happen to figure it out I'll let
you know. In the meantime I'll stick with what I've been using so far.

Thank you.
Best regards,
Eduardo.

drakorg, 2020-07-07:
Hi, I just came across this, a post by @mdegans where he builds a gstreamer pipeline with rtsp authenticated access. I guess this is what you were asking for?

https://forums.developer.nvidia.com/t/jetson-inference-python-multiprocessing/78291/11

Regards,
Eduardo

mdegans, 2020-07-07:
> Hi, I just came across this, a post by @mdegans where he builds a gstreamer pipeline with rtsp authenticated access. I guess this is what you were asking for?
> 
> https://forums.developer.nvidia.com/t/jetson-inference-python-multiprocessing/78291/11
> 
> Regards,
> Eduardo

@drakorg 

I think that post might be incorrect. I was quoting from another post, but looking at the documentation for rtspsrc It's unclear whether that syntax works for specifying the creds.

https://gstreamer.freedesktop.org/documentation/rtsp/rtspsrc.html#rtspsrc:user-id

In any case, it's not really a good idea to specify a password at the command prompt or in a URI. Also, RTSP does not encrypt the credentials by default. You need to set up TLS for that.

dusty-nv, 2020-07-07:
@mdegans I have seen a couple other examples online of it specified as `rtspsrc location=rtsp://username:password@ip:port`, so I will try both ways and see what works.  It would simplify my parsing if your way worked - perhaps rtsp parses it internally.

dusty-nv, 2020-07-07:
It does appear that the `rtsp://username:password@ip:port` form works fine, yay.

@drakorg if you pull the latest from dev branch, you should be able to test it.

The way I tested it was with the `test-auth` server example from this repo:  https://github.com/GStreamer/gst-rtsp-server

mdegans, 2020-07-07:
> @mdegans I have seen a couple other examples online of it specified as `rtspsrc location=rtsp://username:password@ip:port`, so I will try both ways and see what works. It would simplify my parsing if your way worked - perhaps rtspsrc element parses it internally.

@dusty-nv 
I haven't looked at the source for that element, but GStreamer has some url parsing stuff that's largely missing from GLib itself in case you ever need something like python's urlparse in GObject.

https://gstreamer.freedesktop.org/documentation/gstreamer/gsturihandler.html?gi-language=c#gst_uri_new

Just tried it in Python and it does appear it expects the userinfo part in the beginning:
```python
>>> uri = Gst.Uri.new('http', 'foo:bar', 'host', 42, '/some/path', None, None)
>>> uri.get_userinfo()
>>> 'foo:bar'
>>> >>> uri.to_string()
'http://foo:bar@host:42/some/path'
>>> uri = Gst.Uri.from_string('http://user:pass@host:42/some/path')
>>> uri.get_userinfo()
'user:pass'
>>> Gst.Uri.from_string('rtsp://username:password@ip:port') == None
True
```

drakorg, 2020-07-07:
Hi, sure, I'll try it out.

drakorg, 2020-07-08:
Hi, just tested and I'm still getting the error. 
When I don't supply the credentials, I get the expected Unauthorized error.

```
(nspi) drakorg@drakorg-desktop:~/workspace/nspi$ detectnet-camera "rtsp://192.168.0.22/"
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for 192.168.0.22/
[gstreamer] gstDecoder -- Unauthorized
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://192.168.0.22/
detectnet:  failed to create input stream
```

But when I do, I get the following error instead (same error as I was getting before if recall correctly).

```
(nspi) drakorg@drakorg-desktop:~/workspace/nspi$ detectnet-camera "rtsp://admin:xxxxr@192.168.0.22/"
URI -- failed to parse IP port from rtsp://admin:xxxx@192.168.0.22/
[video]  videoOptions -- failed to parse input resource URI (rtsp://admin:xxxx@192.168.0.22/)
[video]  videoSource -- failed to parse command line options
detectnet:  failed to create input stream

```
I tested this on a fresh clone of the dev branch just in case.

dusty-nv, 2020-07-08:
Can you try specifying the port in the URL?     ie `rtsp://admin:xxxx@192.168.0.22:1234`  (where 1234 is the port number)

drakorg, 2020-07-08:
YES. 👍 😄

dusty-nv, 2020-07-08:
OK, thanks.  I will check the support for when a port isn't explicitly specified then.

For my info, when the port isn't specified, what is the default port for RTSP?   Is it 554?

drakorg, 2020-07-08:
554 yes.

drakorg, 2020-07-08:
I'll have to get familiar with this new pipeline now, but I think it's the way to go. Best thing if this probes performant, I will finally be able to get rid of my current pipeline based on ffmpeg + h264_nvmpi + v4l2, which required many different components working together to be able to use hardware h264 decoding. Also, I was having some problems with ffmpeg process stopping under certain conditions which I haven't been able to identify yet (it may probably be related to overheating when running several channels and my lack of any cooling mechanism for the moment).

Finally, can I throttle fps and select a custom size of the fetched frames under gstreamer? Almost 100% sure that it can be done, but just to be sure. I'll have to follow some tutorial on the whole gst thing, but I've renewed hopes thanks to this.

Thanks a lot.

dusty-nv, 2020-07-08:
> 554 yes.

OK, I have added code that defaults the RTSP port to 554 if it isn't specified.   If you pull the latest from the dev branch, and it doesn't run with `rtsp://admin:xxxxr@192.168.0.22/` (i.e. not specifying the RTSP port), please let me know.

> can I throttle fps and select a custom size of the fetched frames under gstreamer?

Yes, if you launch the program with the `--input-width=X`, `--input-height=Y`, and `--input-rate=Z` options, you can set custom resolution / framerate.   Note that I just added the support for custom `--input-rate`, so you would need to pull the latest for that.

drakorg, 2020-07-11:
Hi, the rtsp port now defaults to 554 when not specified explicitly.
Thank you.

closed 2020-07-11 by drakorg