dusty-nv/jetson-utils #28: An issue on a video encoding

rose-jinyang, 2020-07-07:
Hello
How are you?
Thanks for contributing this project.
I've tried to encode a H264 video with this library on Jetson.
I set videoOpetion as the following.
videoOptions video_option;
				_video_option.resource = "file:///" + input_video_path;
				video_option.width = in_w;
				video_option.height = in_h;
				video_option.frameRate = fps;
				video_option.zeroCopy = true;
				video_option.deviceType = videoOptions::DEVICE_FILE;
				video_option.ioType = videoOptions::OUTPUT;
				video_option.codec = videoOptions::CODEC_H264;
				input_writer = gstEncoder::Create(video_option);
				input_writer->Open();_

Then, the redering is following.
                                _input_writer->Render(img.data, img.cols, img.rows, IMAGE_RGB8);_

But when this code is run, the following issue occurs.

_[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/wakesys/Jin/jetson-utils/cuda/cudaYUV-YV12.cu:257
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/wakesys/Jin/jetson-utils/cuda/cudaColorspace.cpp:128
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /home/wakesys/Jin/jetson-utils/codec/gstEncoder.cpp:568
[gstreamer] gstEncoder::Render() -- unsupported image format (rgb8)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f_

So the video encoding is failed.
Please let me know any solution.
Thanks

dusty-nv, 2020-07-07:
It looks like this memory wasn't allocated in CUDA memory.  First allocate a buffer with cudaAllocMapped() from cudaMappedMemory.h, then memcpy your buffer to it.  Then pass the CUDA buffer to Render() instead of your original one.

rose-jinyang, 2020-07-07:
Hi
Thanks for your quick reply.
Your thought is right.
I have one more question.
I am trying to encode two video files.
so I created two gstEncoder instances.
The first video encoder works well but the second does not.
What do u think the reason is?
Thanks

rose-jinyang, 2020-07-08:
Oh, I've fixed this.

closed 2020-07-08 by rose-jinyang