dusty-nv/jetson-utils #30: gstreamer stalling?

drakorg, 2020-07-11:
Hi, I'm having a problem while running an RTSP videoSource from the dev branch (the one that allows for authenticated uris). It happens from time to time (maybe 1/3 of the runs) ... the thing is that the videoSource.Capture() method kinda stalls, waiting for some condition on the gstreamer pipeline to finish, and while it cant resolve what's going on in the pipeline, it would appear as if it didn't return ...

This is part of the gstreamer output that I see while this condition is active ... and while this happens my program basically hangs waiting for something that apparently never happens ...

[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> videorate3
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter18
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv3
[gstreamer] gstreamer changed state from NULL to READY ==> omxh264dec-omxh264dec8
[gstreamer] gstreamer changed state from NULL to READY ==> h264parse18
[gstreamer] gstreamer changed state from NULL to READY ==> rtph264depay18
[gstreamer] gstreamer changed state from NULL to READY ==> queue8
[gstreamer] gstreamer changed state from NULL to READY ==> rtspsrc8
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline8
[gstreamer] gstreamer changed state from READY to PAUSED ==> videorate3
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter18
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv3
[gstreamer] gstreamer changed state from READY to PAUSED ==> omxh264dec-omxh264dec8
[gstreamer] gstreamer changed state from READY to PAUSED ==> h264parse18
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtph264depay18
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> queue8
[gstreamer] gstreamer message progress ==> rtspsrc8
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtspsrc8
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline8
[gstreamer] gstreamer message new-clock ==> pipeline8
[gstreamer] gstreamer message progress ==> rtspsrc8
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> videorate3
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter18
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv3
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> omxh264dec-omxh264dec8
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> h264parse18
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtph264depay18
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> queue8
[gstreamer] gstreamer message progress ==> rtspsrc8
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtspsrc8
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)11583360, maximum-bitrate=(uint)32654560, bitrate=(uint)15395729;
[gstreamer] gstreamer message progress ==> rtspsrc8
[gstreamer] gstreamer rtspsrc8 ERROR Could not read from resource.
[gstreamer] gstreamer Debugging info: gstrtspsrc.c(5917): gst_rtsp_src_receive_response (): /GstPipeline:pipeline8/GstRTSPSrc:rtspsrc8:
Could not receive message. (System error)
[gstreamer] gstreamer message progress ==> rtspsrc8
[gstreamer] gstreamer message progress ==> rtspsrc8
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)2345800, maximum-bitrate=(uint)20655400, bitrate=(uint)3813294;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)7878000, maximum-bitrate=(uint)33955800, bitrate=(uint)11630612;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)2345800, maximum-bitrate=(uint)20690000, bitrate=(uint)3869547;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)2342200, maximum-bitrate=(uint)20690000, bitrate=(uint)3807197;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)11583360, maximum-bitrate=(uint)32654560, bitrate=(uint)15721958;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)7878000, maximum-bitrate=(uint)33955800, bitrate=(uint)11870816;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)2336400, maximum-bitrate=(uint)20690000, bitrate=(uint)3815523;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)2326800, maximum-bitrate=(uint)20690000, bitrate=(uint)3808952;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)884400, maximum-bitrate=(uint)33955800, bitrate=(uint)11931955;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)870800, maximum-bitrate=(uint)33955800, bitrate=(uint)11945827;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)749800, maximum-bitrate=(uint)33955800, bitrate=(uint)11903179;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)166000, maximum-bitrate=(uint)33955800, bitrate=(uint)11890159;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)2326800, maximum-bitrate=(uint)20690600, bitrate=(uint)3813485;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)2326800, maximum-bitrate=(uint)20693000, bitrate=(uint)3813473;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)2313800, maximum-bitrate=(uint)20693000, bitrate=(uint)3806887;

I've already learned to recognize this pattern as a "ok, ill have to kill it and run it again" ... it's like if gstreamer has some problem starting the pipeline ... but it doesnt give me the chance to reconnect or anything.

I can attach the whole session logs if you need them.
Thank you.

dusty-nv, 2020-07-12:
Sometimes with RTP / RTSP it can take some time to "connect" with the remote source over the network and actually start streaming.  In my experience up to 10 seconds or so.

If you just let it run, does it ever give you back a frame or not?

drakorg, 2020-07-12:
Hi, yes, I'm used to dealing with those kinds of delays, glitches,
disconnections or even periods of no communication. That's the reason the
fetch loop is always assuming that a problem can occur, and in that case it
can retry or even restart the channel. The problem with what I'm reporting
is that the call never returns, and I have nothing else to do but to kill
the application. When this happens the thread stalls and the only logging
that the application receives are a never ending sequence of this:

[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ (Main\
Profile)", minimum-bitrate=(uint)2326800, maximum-bitrate=(uint)20690600,
bitrate=(uint)3813485;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ (Main\
Profile)", minimum-bitrate=(uint)2326800, maximum-bitrate=(uint)20693000,
bitrate=(uint)3813473;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ (Main\
Profile)", minimum-bitrate=(uint)2313800, maximum-bitrate=(uint)20693000,
bitrate=(uint)3806887;

If you need any more information please say and I'll try to prepare it.
Thank you.

dusty-nv, 2020-07-12:
Hmm I see this error in the log:

GstPipeline:pipeline8/GstRTSPSrc:rtspsrc8:
Could not receive message. (System error)

So when you call videoSource.Capture(), set timeout=1000, and it will return after one second.  src.Capture(timeout=1000)

drakorg, 2020-07-12:
Hi, thanks for the suggestion. However I get the following error when passing the timeout parameter to Capture().

```
 2020-07-12 12:31:55,461 [DEBUG] threadId=546567373296: about to Capture(timeout=1000) .. 
 2020-07-12 12:31:55,463 [DEBUG] threadId=546567373296: Capture() finished ... 
 2020-07-12 12:31:55,464 [WARNING] threadId=546567373296: jetson.utils -- videoSource.Capture() failed to parse args tuple
```
Just in case the snippet:

```
    def fetch_frame(self):
        timeout = 1000
        self.debug('about to Capture(timeout={}) ..'.format(timeout))
        try:
            cudaImage = self.stream.Capture(timeout = timeout)
        finally:
            self.debug('Capture() finished ...')
```

dusty-nv, 2020-07-12:
Sorry about that, it was a bug - just fixed it.  It is working here now.

Try pulling the latest on jetson-inference (`git pull origin dev`) and it should be working now.

drakorg, 2020-07-13:
Hi, the timeout works now, but the problem kinda remains.
I'm now able to reproduce it at least. Apparently the problem is not in the Capture() method, but in the Close() one.

I'm attaching the full log but the important lines are mostly the last ones, where Close() is called and it never comes back.

```
(nspi) drakorg@drakorg-desktop:~/workspace/nspi$ python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import jetson.utils
>>> import sys
>>> vs = jetson.utils.videoSource('rtsp://admin:xxxx@192.168.0.20', sys.argv + ['--headless=true'])
URI -- missing/invalid IP port from rtsp://admin:xxxx@192.168.0.20, default to port 554
[gstreamer] gstDecoder -- creating decoder for admin
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- discovered video resolution: 1920x1080  (framerate 25.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)4.2, profile=(string)main, width=(int)1920, height=(int)1080, framerate=(fraction)25/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://admin:xxxx@192.168.0.20 ! queue ! rtph264depay ! h264parse ! omxh264dec ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from rtsp://admin:xxxx@192.168.0.20
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://admin:xxxx@192.168.0.20
     - protocol:  rtsp
     - location:  admin
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      h264
  -- width:      1920
  -- height:     1080
  -- frameRate:  25.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
>>> while True:
...     frame = vs.Capture(timeout = 1000)
...
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter2
[gstreamer] gstreamer changed state from NULL to READY ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from NULL to READY ==> h264parse2
[gstreamer] gstreamer changed state from NULL to READY ==> rtph264depay2
[gstreamer] gstreamer changed state from NULL to READY ==> queue0
[gstreamer] gstreamer changed state from NULL to READY ==> rtspsrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter2
[gstreamer] gstreamer changed state from READY to PAUSED ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from READY to PAUSED ==> h264parse2
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtph264depay2
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> queue0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtspsrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter2
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> h264parse2
[... 190 lines cut ...]
[gstreamer] gstDecoder -- end of stream (EOS)

>>> vs.IsStreaming()
False
>>> vs.Open()
[gstreamer] gstDecoder -- end of stream (EOS) has been reached, stream has been closed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
Exception: jetson.utils -- failed to open videoSource device for streaming
>>> vs.Capture()
[gstreamer] gstDecoder -- end of stream (EOS) has been reached, stream has been closed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
Exception: jetson.utils -- videoSource failed to capture image
>>> vs.Close()
[gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL

^C

```
And then I had to kill the interpreter.

Apparently Close() is never getting back. I would expect Close() to return an error if it's not able to close its underlying stuff (because it reached EOS and it was already closed, for example), or maybe just do nothing and just return, but I'd never expect it to block indefinitely. That I think is another manifestation of the same problem I was facing before (or the very same problem or root of it). 

Thank you.

drakorg, 2020-07-13:
Hi, the same happens also when removing the last reference to a videoSource which has internally been closed (or stopped, or has received an EOS). It hangs on the sole dereference of the object (vs = None). Check the last line. The interpreter never comes back after that line.

```
(nspi) drakorg@drakorg-desktop:~/workspace/nspi$ python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import jetson.utils
>>> import sys
>>> vs = jetson.utils.videoSource('rtsp://admin:xxxx@192.168.0.20', sys.argv + ['--headless=true'])
URI -- missing/invalid IP port from rtsp://admin:xxxx@192.168.0.20, default to port 554
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin
nvbuf_utils: Could not get EGL display connection
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:xxxx@192.168.0.20
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
Exception: jetson.utils -- failed to create videoSource device
>>> vs = jetson.utils.videoSource('rtsp://admin:xxxx@192.168.0.20', sys.argv + ['--headless=true'])
URI -- missing/invalid IP port from rtsp://admin:xxxx@192.168.0.20, default to port 554
[gstreamer] gstDecoder -- creating decoder for admin
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- discovered video resolution: 1280x720  (framerate 25.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)main, width=(int)1280, height=(int)720, framerate=(fraction)25/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://admin:xxxx@192.168.0.20 ! queue ! rtph264depay ! h264parse ! omxh264dec ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from rtsp://admin:xxxx@192.168.0.20
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://admin:xxxx@192.168.0.20
     - protocol:  rtsp
     - location:  admin
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      h264
  -- width:      1280
  -- height:     720
  -- frameRate:  25.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
>>> vs = jetson.utils.videoSource('rtsp://admin:xxxx@192.168.0.20', sys.argv)
URI -- missing/invalid IP port from rtsp://admin:xxxx@192.168.0.20, default to port 554
[gstreamer] gstDecoder -- creating decoder for admin
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- discovered video resolution: 1280x720  (framerate 25.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)main, width=(int)1280, height=(int)720, framerate=(fraction)25/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://admin:xxxx@192.168.0.20 ! queue ! rtph264depay ! h264parse ! omxh264dec ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from rtsp://admin:xxxx@192.168.0.20
------------------------------------------------
gstDecoder video options:
------------------------------------------------
[... 234 lines cut ...]
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)1688000, maximum-bitrate=(uint)17720800, bitrate=(uint)3177934;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)1688000, maximum-bitrate=(uint)17720800, bitrate=(uint)3106318;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)1688000, maximum-bitrate=(uint)17720800, bitrate=(uint)3041829;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)1688000, maximum-bitrate=(uint)17720800, bitrate=(uint)3155077;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)1688000, maximum-bitrate=(uint)17720800, bitrate=(uint)3082528;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)1688000, maximum-bitrate=(uint)17720800, bitrate=(uint)3016443;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)1688000, maximum-bitrate=(uint)17720800, bitrate=(uint)3137671;
[... cut ...]
Allocating new output: 1920x1088 (x 11), ThumbnailMode = 0
OPENMAX: HandleNewStreamFormat: 3605: Send OMX_EventPortSettingsChanged: nFrameWidth = 1920, nFrameHeight = 1080
[gstreamer] gstDecoder -- end of stream (EOS)
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
Exception: jetson.utils -- videoSource failed to capture image
>>> vs.IsStreaming()
False
>>> vs = None
[gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL

^C

```

Thank you.
Best regards.

dusty-nv, 2020-07-13:
Hmm it appears that GStreamer never returns from `gst_element_set_state(mPipeline, GST_STATE_NULL);`

It seems like it is blocking on network access:   https://gstreamer-devel.narkive.com/cePG8Gw8/gst-element-set-state-hangs#post2

I will have to think of a way to workaround it.   For now, what happens if you comment out this line of code:

https://github.com/dusty-nv/jetson-utils/blob/5cf543d9e0bc37ce7df0def2242c2e17dc8559d0/codec/gstDecoder.cpp#L114

Then run `make`, and `sudo make install` again.

drakorg, 2020-07-13:
Hi, great, apparently that worked, it no longer hangs and I can now do the reconnection attempt without problem apparently. I'll let you know if I face some long term issue, but for the moment apparently it's doing well.
Thank you,
Regards.

yashjal, 2020-11-04:
@dusty-nv, I am facing a similar issue while streaming 120 fps videos from 2 CSI cameras on the jetson nano using the C++ API. The Capture() call times out for one of the cameras/streams after a while of streaming (maybe 5-10 mins) and then when I do a SAFE_DELETE() of that input stream, it hangs after ```[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL``` at which point I have to force kill the program. I'm not sure if the problem arises from the same place, i.e. at ```gst_element_set_state(mPipeline, GST_STATE_NULL);```. I will try and use the suggestion of commenting out the Close() and inform back.

elcapitan26, 2020-12-28:
@dusty-nv I've been following your S3E1 Hello AI World tutorial and ran into an issue that could be related to this thread. I have a live RTSP feed that I can play on network and can view on the Jetson Nano using the pre-installed SMPlayer. When I feed the path into the video-viewer app it attempts to pull the stream but eventually fails in a loop and I have to kill it. This happens with docker and source versions. The //close() suggestion above did not help. 

My server sees the connection attempt with a disconnect issued one second later. I'm thinking this is a timing issue caused by my network but any work-around would be appreciated. (Thanks for the tutorials by the way. They are great.)

[Start of the failure loop:]
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
video-viewer:  failed to capture video frame
video-viewer:  failed to capture video frame
video-viewer:  failed to capture video frame

hakao32, 2021-06-19:
I also have same issue. And //close did not help me either.

dusty-nv, 2021-06-21:
Are you able to create a standalone GStreamer pipeline for viewing your RTSP feed and run it with gst-launch-1.0?   This would be good for testing purposes to verify that GStreamer can indeed pull your RTSP feed.

NISHANTSHRIVASTAV, 2021-08-04:
Hi @dusty-nv  

We are able to do object detection with RTSP using detectnet utility on Jetson Xavier NX but when we are doing object detection using python jetson.inference and jetson.utils modules it is giving us the above issue. 

We have referred to the master branch implementation and it is giving us the same issue:

```
(python3:20080): GStreamer-CRITICAL **: 04:07:32.972: gst_caps_is_empty: assertion 'GST_IS_CAPS (caps)' failed

(python3:20080): GStreamer-CRITICAL **: 04:07:32.972: gst_caps_truncate: assertion 'GST_IS_CAPS (caps)' failed

(python3:20080): GStreamer-CRITICAL **: 04:07:32.972: gst_caps_fixate: assertion 'GST_IS_CAPS (caps)' failed

(python3:20080): GStreamer-CRITICAL **: 04:07:32.972: gst_caps_get_structure: assertion 'GST_IS_CAPS (caps)' failed

(python3:20080): GStreamer-CRITICAL **: 04:07:32.972: gst_structure_get_string: assertion 'structure != NULL' failed

(python3:20080): GStreamer-CRITICAL **: 04:07:32.972: gst_mini_object_unref: assertion 'mini_object != NULL' failed
```
**Is it stable now in the master branch?** 

Please let us know if we could provide more information.

NISHANTSHRIVASTAV, 2021-08-05:
Hello @dusty-nv,

We are able to do object detection using detectnet/detectnet.py or video-viewer/video-viewer.py but faced the above issue while using jetson inference python API.

We have used the timeout argument i.e` input.Capture(timeout=15000)` and it was working fine.

dusty-nv, 2021-08-05:
> We are able to do object detection using detectnet/detectnet.py or video-viewer/video-viewer.py but faced the above issue while working with RTSP stream with jetson inference python API.

detectnet.py / video-viewer.py use the same jetson.inference Python API, so if you start from those examples, it should still be working.

> We have used the timeout argument i.e` input.Capture(timeout=15000)` and it was working fine.

Do you mean that it is working fine for you now with `timeout=15000`?

NISHANTSHRIVASTAV, 2021-09-02:
> detectnet.py / video-viewer.py use the same jetson.inference Python API, so if you start from those examples, it should still be working.
>
>
> Do you mean that it is working fine for you now with [...]

Yup, we are checking whether we are getting input frames in between 15 sec and if not we are rebooting the Jetson.

jwrl7, 2022-01-23:
Has there been any solution to this?
I have a similar  issue and running into stability problems while trying to start and stop the video-viewer application. If i have to restart video-viewer for any reason, I must reboot the xavier. Iv also noticed that letting the application run for 12+ hours the frame rates degrades due to increased capture timeouts being issued. Im not sure if this is a gstreamer thing or nvargus-daemon thing.

rajeshroy402, 2022-02-04:
Hey @dusty-nv , 
I am on the latest origin and have checkout to nvmm-disabled. Still getting this error

`Allocating new output: 1280x720 (x 16), ThumbnailMode = 0
OPENMAX: HandleNewStreamFormat: 3605: Send OMX_EventPortSettingsChanged: nFrameWidth = 1280, nFrameHeight = 720
[gstreamer] gstDecoder -- end of stream (EOS)`

dusty-nv, 2022-02-04:
> Allocating new output: 1280x720 (x 16), ThumbnailMode = 0 OPENMAX: HandleNewStreamFormat: 3605: Send OMX_EventPortSettingsChanged: nFrameWidth = 1280, nFrameHeight = 720 [gstreamer] gstDecoder -- end of stream (EOS)

That isn't an error persay (well, at least not from jetson.utils) - it gets triggered when the stream source ends.  Are you able to play the stream in a standalone GStreamer pipeline to check that it's working?