dusty-nv/jetson-utils #31: headless mode by --headless=true

drakorg, 2020-07-13:
Hi, I believe that the correct way to set the --headless mode is by passing --headless=true instead of --headless alone as I've seen in one of the examples (detectnet-camera at least).

Here is an example session where you can see --headless=true working, while --headless alone fails.

```
drakorg@drakorg-desktop:~/workspace/nspi$ python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> import jetson.utils
>>> vs = jetson.utils.videoSource('rtsp://admin:xxxx@192.168.0.20', sys.argv + ['--headless'])
URI -- missing/invalid IP port from rtsp://admin:xxxx@192.168.0.20, default to port 554
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin
nvbuf_utils: Could not get EGL display connection
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:xxxx@192.168.0.20
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
Exception: jetson.utils -- failed to create videoSource device
>>> vs = jetson.utils.videoSource('rtsp://admin:xxxx@192.168.0.20', sys.argv + ['--headless=true'])
URI -- missing/invalid IP port from rtsp://admin:xxxx@192.168.0.20, default to port 554
[gstreamer] gstDecoder -- creating decoder for admin
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- discovered video resolution: 1280x720  (framerate 25.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)main, width=(int)1280, height=(int)720, framerate=(fraction)25/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://admin:xxxx@192.168.0.20 ! queue ! rtph264depay ! h264parse ! omxh264dec ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from rtsp://admin:xxxx@192.168.0.20
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://admin:xxxx@192.168.0.20
     - protocol:  rtsp
     - location:  admin
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      h264
  -- width:      1280
  -- height:     720
  -- frameRate:  25.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
>>>

```

That's the reason detectnet-camera example does not work from commandline despite specifying the headless switch, as seen below:

```
drakorg@drakorg-desktop:~/workspace/nspi/scripts/jetson-nano/jetson-inference$ detectnet-camera "rtsp://admin:xxxx@192.168.0.20" --headless
URI -- missing/invalid IP port from rtsp://admin:xxxx@192.168.0.20, default to port 554
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin
nvbuf_utils: Could not get EGL display connection
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:xxxx@192.168.0.20
detectnet:  failed to create input stream
drakorg@drakorg-desktop:~/workspace/nspi/scripts/jetson-nano/jetson-inference$ detectnet-camera "rtsp://admin:xxxx@192.168.0.20" --headless=true
URI -- missing/invalid IP port from rtsp://admin:xxxx@192.168.0.20, default to port 554
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin
nvbuf_utils: Could not get EGL display connection
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:xxxx@192.168.0.20
detectnet:  failed to create input stream
```

Now, if you see how detectnet-camera handles the switch, it only adds "--headless" to the argv (under some condition), while it should be "--headless=true".

```
drakorg@drakorg-desktop:~/workspace/nspi/scripts/jetson-nano/jetson-inference$ cat python/examples/detectnet.py
#!/usr/bin/python3
#
# Copyright (c) 2020, NVIDIA CORPORATION. All rights reserved.
#
# Permission is hereby granted, free of charge, to any person obtaining a
# copy of this software and associated documentation files (the "Software"),
# to deal in the Software without restriction, including without limitation
# the rights to use, copy, modify, merge, publish, distribute, sublicense,
# and/or sell copies of the Software, and to permit persons to whom the
# Software is furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.  IN NO EVENT SHALL
# THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
# FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
# DEALINGS IN THE SOFTWARE.
#

import jetson.inference
import jetson.utils

import argparse
import sys

# parse the command line
parser = argparse.ArgumentParser(description="Locate objects in a live camera stream using an object detection DNN.",
                                 formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.detectNet.Usage() +
                                 jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("input_URI", type=str, default="", nargs='?', help="URI of the input stream")
parser.add_argument("output_URI", type=str, default="", nargs='?', help="URI of the output stream")
parser.add_argument("--network", type=str, default="ssd-mobilenet-v2", help="pre-trained model to load (see below for options)")
parser.add_argument("--overlay", type=str, default="box,labels,conf", help="detection overlay flags (e.g. --overlay=box,labels,conf)\nvalid combinations are:  'box', 'labels', 'conf', 'none'")
parser.add_argument("--threshold", type=float, default=0.5, help="minimum detection threshold to use")
parser.add_argument("--width", type=int, default=1280, help="desired width of camera stream (default is 1280 pixels)")
parser.add_argument("--height", type=int, default=720, help="desired height of camera stream (default is 720 pixels)")

is_headless = ["--headless"] if sys.argv[0].find('console.py') != -1 else [""]

try:
        opt = parser.parse_known_args()[0]
except:
        print("")
        parser.print_help()
        sys.exit(0)

# load the object detection network
net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)

# create video sources & outputs
input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
output = jetson.utils.videoOutput(opt.output_URI, argv=sys.argv+is_headless)

# process frames until the user exits
while True:
        # capture the next image
        img = input.Capture()

        # detect objects in the image (with overlay)
        detections = net.Detect(img, overlay=opt.overlay)

        # print the detections
        print("detected {:d} objects in image".format(len(detections)))

        for detection in detections:
                print(detection)

        # render the image
        output.Render(img)

        # update the title bar
        output.SetStatus("{:s} | Network {:.0f} FPS".format(opt.network, net.GetNetworkFPS()))

        # print out performance info
        net.PrintProfilerTimes()

        # exit on input/output EOS
        if not input.IsStreaming() or not output.IsStreaming():
                break

```

Thank you.
Regards.

dusty-nv, 2020-07-13:
Hmm what if you set --headless first on the command line, before the positional args (i.e. the URI)

drakorg, 2020-07-13:
Hi, please disregard the issue. Apparently it's related to the first attempt always failing and the second one always working (in my case), despite the value or presence of the switch, which on the other hand makes more sense since we were talking about the videoSource and not the videoOutput. Maybe there's something odd on my installation and/or camera. I will check later, but it's not related to the headless parameter apparently.

Thank you.
Regards.

closed 2020-07-13 by drakorg