dusty-nv/jetson-utils #32: gstreamer rtsp lagging badly on nano

drakorg, 2020-07-13:
Hi, the new videoSource with the gstreamer backend for rtsp streams appears to be lagging badly (up to 30 secs) on a single channel while the nano is sitting idle except for this. I can see it lag even with the camera-viewer stock app. I will update with more information when I have more time, but I wanted to open the issue not to forget about it. With this level of lag I'll be forced again to move away from this component. I'm running Jetson 4.3 without any package updates.

Regards.
Eduardo.

dusty-nv, 2020-07-13:
If you just use GStreamer standalone (i.e. `gst-launch-1.0` with `xvimagesink` instead of `appsink` like I use), is there still the latency?

I see suggestions from below to set `latency=0` in the `rtspsrc` element and to add `queue` elements to the pipeline:   https://stackoverflow.com/q/42695673

Also, can you determine if the latency goes away or decreases if you set the camera to a lower resolution?

drakorg, 2020-07-15:
Hi, I just made some tests, and the detectnet-camera demo app at 1920x1080 only lags behind from 2 to 3 seconds, but at least its constant and doesn't increase, which is fine and acceptable. The problem comes with bigger frame sizes such as 2688x1520 (one of my cams has that maximum resolution) ... in that case the demo-app lags one second for every second ... meaning that in 30 seconds of live running the video feed will lag 30 seconds ... which is totally unacceptable. We can then discuss if that's a resolution worth worrying for, I agree, but at least it serves a reference. I will now try to resize the stream via --width parameters to detectnet-camera and see what happens. Will update soon.

drakorg, 2020-07-15:
Regarding my previous comment, about feeding a larger frame in an attempt to pick small details: do I actually gain anything by feeding a larger frame to detectnet (ssd-mobilenet-v2 for instance, or for any other network) or will it be sized down for the detection anyways? If that's the case, what resolution would be the biggest that it would make sense to pass it? If that was the case I guess it could also vary from network to network, and if so, would there be any table stating the max resolutions that would make sense to pass to them?

drakorg, 2020-07-15:
> If you just use GStreamer standalone (i.e. `gst-launch-1.0` with `xvimagesink` instead of `appsink` like I use), is there still the latency?
>
> I see suggestions from below to set `latency=0` in the [...]

Hi, I'm not very familiar with gstreamer yet, could you please provide me the exact pipeline (and command line) you would like me to try? I tried with some variations of the pipeline that was printed by the detectnet-camera application but I'm getting nothing but errors.

Thank you.

dusty-nv, 2020-07-15:
> o I actually gain anything by feeding a larger frame to detectnet (ssd-mobilenet-v2 for instance, or for any other network) or will it be sized down for the detection anyways?

It gets sized down to 300x300 anyways inside of detectNet (before passing it to SSD-Mobilenet), so it's not really worth it to use high-resolution unless you need it for other things too.

Does it help if you add a `queue` element at this line?

https://github.com/dusty-nv/jetson-utils/blob/4438da95d93ea87aa55031695d30aaac6bdac883/codec/gstDecoder.cpp#L615

Add this code at that line:  `ss << " ! queue ! ";`

Can you also try un-commenting this line?

https://github.com/dusty-nv/jetson-utils/blob/4438da95d93ea87aa55031695d30aaac6bdac883/codec/gstDecoder.cpp#L521
You can mess with the latency (i.e. change it to 0)

When done editing, then re-run `make` and `sudo make install` to test it.
Unfortunately I can't seem to reproduce this problem, I will have to get a real RTSP camera or something.  Which one do you use?

drakorg, 2020-07-15:
> It gets sized down to 300x300 anyways inside of detectNet (before passing it to SSD-Mobilenet), so it's not really worth it to use high-resolution unless you need it for other things too.

Ok, thanks for the clarification.

> Does it help if you add a `queue` element at this line?
>
> https://github.com/dusty-nv/jetson-utils/blob/4438da95d93ea87aa55031695d30aaac6bdac883/codec/gstDecoder.cpp#L615
>
> Add this code at that line: [...]

I will try as soon as I can.

> Unfortunately I can't seem to reproduce this problem, I will have to get a real RTSP camera or something. Which one do you use?

For testing purposes I'd recommend this one:
https://www.amazon.com/Hikvision-DS-2CD2442FWD-IW-Indoor-Wireless-Camera/dp/B071HM1FDB
Since it's an indoor camera that you can have next to you. This particular model supports the resolution I mentioned before, so you should be able to reproduce it.

Thank you.
Regards.
Eduardo