dusty-nv/jetson-utils #49: Multiple glDisplay streams error

adryash, 2021-02-09:
Hello! 
I am trying to implement multiple camera streams and show them on one display using glDisplay class. There are no problems when I have one stream, but when I start displaying the second one I have segmentation fault on my Jetson Nano after 1-3 frames. I think that it may have something to do with the multi-threading - every camera has its own thread and they are being processes and send to videoOutput simultaneously. Any advice? Is it possible to have multiple streams displayed in real time on my Jetson Nano at all?

Here is part of my code in the Camera class:
```
videoOutput* m_pScreenOutput = videoOutput::Create(0, nullptr);
cv::cuda::GpuMat	m_gmImgDisplay;	

// streaming loop
{
  // process the frame and put it into m_gmImgDisplay
  m_pScreenOutput->Render((uchar4*) m_gmImgDisplay.data, m_uWidth,  m_uHeight, IMAGE_RGBA8);
  m_pScreenOutput->SetStatus(m_acDevName);
}

```

dusty-nv, 2021-02-09:
As a test, if you remove the threading, does it still crash?  

Others have used multiple glDisplay windows.  Or you could create one window and use cudaOverlay to compost the independent frames into one larger image and just render it once.

adryash, 2021-02-11:
Thanks for the response!

> As a test, if you remove the threading, does it still crash?
No, displaying two cameras in single thread sequentially works fine. 

> Or you could create one window and use cudaOverlay to compost the independent frames into one larger image and just render it once.
Yes, it probably would work, but we want to have them separately.