dusty-nv/jetson-utils #53: Problem?: use gstEncoder with rtmp push

jcyhcs, 2021-03-01:
hi,professor:

i use gstEncoder API to encode opencv Mat frame , then push stream to rtmp server with URI like this "rtmp://192.168.1.102:1935/live/livestream", my init code like this:

[image]
when i run my program, i get an error:
[image]
then ,why the rtmp uri become short? just have ip, "live/livestream" has gone! why? 
PLEASE HELP!

dusty-nv, 2021-03-01:
Hi @jcyhcs, I think you need to change this line of `jetson-inference/utils/codec/gstEncoder.cpp:401`:

https://github.com/dusty-nv/jetson-utils/blob/1f3709f48258c2d75500c35605e8f6f4a3447afc/codec/gstEncoder.cpp#L401

to the following:

``` cpp
ss << uri.string << " ";
```

@socieboy added the RTMP support and I haven't tried it, so I'm not sure if the error you received is actually related to this or not.

BTW after you make the change above, re-run `make` and `sudo make install`

socieboy, 2021-03-01:
Seems like the pipeline is being created correctly, but in order to go from rtph264pay to rtmpsink we need another element.

**rtmpsink** only **accepts video/x-flv** as source, and **rtph264pay** has **video/x-h264** as sink, we need to use a flvmux before the rmtp sink.

```
rtph264pay -config-interval=1 ! flvmux ! rtmpsink location="rtmo://{SERVER IP}"
```

jcyhcs, 2021-03-02:
hi, @socieboy  
when i use your gstreamer pipeline, i also get an error: 
[image]
when i remove the element "rtph264pay config-interval=1", then got OK!
so ,please tell me, the element is needed? 
thanks

socieboy, 2021-03-02:
@jcyhcs when you said then got OK! you mean you got the pipeline working and streaming to RTMP?

jcyhcs, 2021-03-02:
@socieboy  YES! this pipe line WORK GOOD!
[image]
please check!

socieboy, 2021-03-02:
you are right, we don't need rtph264, that for the other protocol. i will update the code.

dusty-nv, 2021-03-11:
Patched this in https://github.com/dusty-nv/jetson-utils/commit/ebf55857d60ba0be5c655cacf8eb8e9348ace714

closed 2021-03-11 by dusty-nv