dusty-nv/jetson-utils #56: CUDA ERROR when use gstEncoder!please help!

jcyhcs, 2021-03-02:
hi,prrofessor:
i use gstEncoder API to encode OpenCV Mat frame ,then push stream with RTMP, i use gst pipeline :
[image]
and is OK!
then i use Render API to encode a Mat frame , code this:
[image]
then run the program, i got a CUDA ERROR!:
[image]
what's the problem?? and how can i do ?
help me,please!

dusty-nv, 2021-03-02:
Hi jcyhcs, it needs to be a CUDA pointer that you supply to the function, but you supply CPU-only pointer from OpenCV.   Allocate CUDA memory with cudaAllocMapped() from my cudaMappedMemory.h and memcpy() into it. Then use that CUDA pointer.

[image]<https://user-images.githubusercontent.com/34532224/109579898-b2032b00-7b34-11eb-8d1e-dbaa537e3610.png>
[image]<https://user-images.githubusercontent.com/34532224/109580011-dfe86f80-7b34-11eb-8d01-91b9143d9255.png>
[image]<https://user-images.githubusercontent.com/34532224/109580070-fdb5d480-7b34-11eb-934b-aa06d11fa2e7.png>

jcyhcs, 2021-03-02:
hi @dusty-nv , i modify my code like this:
[image]
then CUDA encode ok.
BUT i got other ERROR!:
[image]
why the pipeline FULL? how can i fix it? PLEASE!

lzcchl, 2021-03-12:
hello, have you solved the problem?

lzcchl, 2021-03-12:
I also want to push rtsp stream, however jetson-utils not yet support, but hw-encode in jetson-utils is great to use;

 so I use jetson-utils + rtsp-server, I want to get the buffer which after encoding to H264(H265), do you know how to get?

I guess the buffer which I need is in "gstEncoder.h" that named 'RingBuffer mBufferYUV', but I haven't test it, if you also have same demand we can finish together.

dusty-nv, 2021-03-12:
> I guess the buffer which I need is in "gstEncoder.h" that named 'RingBuffer mBufferYUV', but I haven't test it, if you also have same demand we can finish together.

Hi @lzcchl, the encoded output isn't typically directly accessible in the code, instead GStreamer handles it (i.e. in filesink or rtpsink).   The `mBufferYUV` are the inputs that have been transformed to YUV colorspace before being sent to GStreamer for compression/transport.

Is there a GStreamer-based RTSP server element that could be integrated into the pipeline?   For example, https://github.com/GStreamer/gst-rtsp-server

This would be the ideal way, to have it all stay inside GStreamer.

dusty-nv, 2021-03-16:
> Hi @lzcchl, the encoded output isn't typically directly accessible in the code, instead GStreamer handles it (i.e. in filesink or rtpsink). The `mBufferYUV` are the inputs that have been transformed [...]

closed 2022-05-27 by jcyhcs