dusty-nv/jetson-utils #57: Problem?: i use gstEncoder success, but JTOP show NVENC is OFF,why?

jcyhcs, 2021-03-02:
hi,professor!
i use gstEncoder success encode opencv Mat frame ,and then push with RTMP, but when the program running, i use jtop tool to check HW engine status, i got that:
[image]
why? NVENC not used??
help me! i confused!

jcyhcs, 2021-03-03:
@socieboy  @dusty-nv   please help ,thanks

socieboy, 2021-03-03:
I believe is because the encoder used for H264 si not one of the NVIDIA encoders, it using omxh264enc, if you want to use one of the nvidia encoders you need to change this [line](https://github.com/dusty-nv/jetson-utils/blob/e101638df4e94a9d576a36644eb6c26733163f84/codec/gstEncoder.cpp#L272) for **nvv4l2h264enc** and  re-run _make_ and _sudo make install_.

I'm not sure if i'm correct but give a try and let us know.

closed 2021-11-09 by jcyhcs