dusty-nv/jetson-utils #59: how can i use gstEncoder to push raw h264 with udp?

jcyhcs, 2021-03-16:
hi,professor:
i want to push raw h264 with udp , then how can i do?
i use gst pipeline like this:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=2000000 ! video/x-h264 !   udpsink host=172.16.100.240 port=9999 auto-multicast=true

the program run with no error. BUT!
when i use vlc to play , i got nothing! i use udp debug tool, i can get hex data. so  how can gstEncoder realize this function? please help me!
@socieboy @dusty-nv

dusty-nv, 2021-03-16:
Try using GStreamer on your PC to recieve/view the stream:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#viewing-rtp-remotely

I haven't found VLC to be very reliable with receiving RTP.

jcyhcs, 2021-03-17:
OK,professor,
is there any possible to transport raw h264 stream JUST WITH UDP! even not use RTP ?.  please teach !

dusty-nv, 2021-03-17:
RTP is a packet protocol on top of UDP.  As you can see in the GStreamer pipeline, it is using udpsrc.   I don't know of other way to just transport raw h264 stream and have VLC play it.

closed 2022-05-27 by jcyhcs