dusty-nv/jetson-utils #66: how can i use jetson-utils to open usb camera with mjpeg format? please help

jcyhcs, 2021-04-15:
hi,professor:
i have a usb camera, i want to read frame with jetson-utils, and read format with mjpeg mode, how can i do it ? please help!

jcyhcs, 2021-04-16:
hi.professor:
i modify camera-viewer program like this:
[image]
then i can open mjpeg stream from usb camera. BUT , i look that NVDEC not working! with JTOP
[image]
so it use soft decode , that right? 

then how can i use NVDEC to capture mjpeg stream from V4L2 USB camera??  
PLEASE HELP ME!
@dusty-nv

dusty-nv, 2021-04-16:
HW JPEG decode is disabled because it was bugging out on some cameras.  Other codecs such as H264/H265 still use HW decode.

If you want, you can re-enable it for MJPEG by changing this line back to what is commented out:

https://github.com/dusty-nv/jetson-utils/blob/c373f49cf21ad2cae7e4d7da7c41f4fd6473958f/camera/gstCamera.cpp#L181

Then re-run `make && sudo make install`

jcyhcs, 2021-04-16:
emm...,ok,i will try!
professor:
if  i read frame use jetson-utils, code like this:
[image]
how can i convert "uchar3*" to OpenCV Mat??
i need opencv Mat to do processing, please help!

closed 2022-05-27 by jcyhcs