dusty-nv/jetson-utils #67: Image Crop isn't working

tkislan, 2021-04-25:
Even when using example implementation in `build/aarch64/bin/cuda-examples.py`, it produces invalid images when cropping
Not sure what is this related to, because when tracked back to cuda implementation, it seems, that is just copies the underlying memory, which is just a 1 dimensional array .. so no idea why the images gets messed up like this
```
output[out_y * outWidth + out_x] = input[in_y * inWidth + in_x];
```

[image]

Note:
Copied issue from https://github.com/dusty-nv/jetson-inference/issues/875
as the code is related to this repo

dusty-nv, 2021-04-26:
Hi @tkislan, please see this post for similar image and make sure that your ROI is specified correctly:

https://forums.developer.nvidia.com/t/cudacrop-mangling-my-image/166467/7?u=dusty_nv

Can you also check that this sample code works correctly for you:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#cropping

tkislan, 2021-04-27:
I've managed to get it working .. but there still is a bug

Given that I have an image with dimensions: `(960x643, 3 channels)`
I want to crop it into 2 overlapping squares

Using these 2 rois works
```
(0, 0, 642, 642)
(317, 0, 959, 642)
```

But the target image needs to be 1 pixel smaller on each size
```
[image] saved 'split_img_0.jpg'  (642x642, 3 channels)
[image] saved 'split_img_1.jpg'  (642x642, 3 channels)
```

Correctly I would want to use same roi points to produce 2 `(643x643, 3 channels)` images

But if I increase allocated image size to `643x643` it produces the messed up image that is posted in the above comment
If I change roi points from `642` to `643` it fails with: `ValueError: jetson.utils -- cudaCrop() had an invalid ROI`
Link to if check: https://github.com/dusty-nv/jetson-utils/blob/c373f49cf21ad2cae7e4d7da7c41f4fd6473958f/python/bindings/PyCUDA.cpp#L1145

dusty-nv, 2021-05-12:
> Same problem here i get the message with next output

@luisfelipesayavedra since the detection coordinates are floating point, the width/height of the image is getting calculated incorrectly.  Instead cast the floating-point detection coordinates to int first:

``` python
# get the cropping ROI from the bounding box
crop_roi = (int(detection.Left), int(detection.Top), int(detection.Right), int(detection.Bottom))
              
imgCropped = jetson.utils.cudaAllocMapped(width=crop_roi[2] - crop_roi[0],
                                              height=crop_roi[3] - crop_roi[1],
                                              format=img.format)

      # crop the image
jetson.utils.cudaCrop(img, imgCropped, crop_roi)

      # save the image
jetson.utils.saveImage('images/test/detection_object_{:d}.jpg'.format(idx), imgCropped)
```

tkislan, 2021-05-12:
So it is not the same problem at all