dusty-nv/jetson-utils #68: How to change Camera exposure settings?

jpmorgan983, 2021-05-05:
We are using this library with the Raspberry Pi Camera V2 (IMX219) on a Nano - is there a way to manually set the exposure?

dusty-nv, 2021-05-05:
Hi @jpmorgan983, jetson-utils uses the `nvarguscamerasrc` element in GStreamer to access MIPI CSI camera.  There are some settings in `nvarguscamerasrc` related to exposure / white balance / saturation / ect:

``` bash
$ gst-inspect-1.0 nvarguscamerasrc

  wbmode              : White balance affects the color temperature of the photo
                        flags: readable, writable
                        Enum "GstNvArgusCamWBMode" Default: 1, "auto"
                           (0): off              - GST_NVCAM_WB_MODE_OFF
                           (1): auto             - GST_NVCAM_WB_MODE_AUTO
                           (2): incandescent     - GST_NVCAM_WB_MODE_INCANDESCENT
                           (3): fluorescent      - GST_NVCAM_WB_MODE_FLUORESCENT
                           (4): warm-fluorescent - GST_NVCAM_WB_MODE_WARM_FLUORESCENT
                           (5): daylight         - GST_NVCAM_WB_MODE_DAYLIGHT
                           (6): cloudy-daylight  - GST_NVCAM_WB_MODE_CLOUDY_DAYLIGHT
                           (7): twilight         - GST_NVCAM_WB_MODE_TWILIGHT
                           (8): shade            - GST_NVCAM_WB_MODE_SHADE
                           (9): manual           - GST_NVCAM_WB_MODE_MANUAL
  saturation          : Property to adjust saturation value
                        flags: readable, writable
                        Float. Range:               0 -               2 Default:               1
  exposuretimerange   : Property to adjust exposure time range in nanoseconds
                        Use string with values of Exposure Time Range (low, high)
                        in that order, to set the property.
                        eg: exposuretimerange="34000 358733000"
                        flags: readable, writable
                        String. Default: null
  gainrange           : Property to adjust gain range
                        Use string with values of Gain Time Range (low, high)
                        in that order, to set the property.
                        eg: gainrange="1 16"
                        flags: readable, writable
                        String. Default: null
  ispdigitalgainrange : Property to adjust digital gain range
                        Use string with values of ISP Digital Gain Range (low, high)
                        in that order, to set the property.
                        eg: ispdigitalgainrange="1 8"
                        flags: readable, writable
                        String. Default: null
  aeantibanding       : property to set the auto exposure antibanding mode
                        flags: readable, writable
                        Enum "GstNvArgusCamAeAntiBandingMode" Default: 1, "AeAntibandingMode_Auto"
                           (0): AeAntibandingMode_Off - GST_NVCAM_AEANTIBANDING_OFF
                           (1): AeAntibandingMode_Auto - GST_NVCAM_AEANTIBANDING_AUTO
                           (2): AeAntibandingMode_50HZ - GST_NVCAM_AEANTIBANDING_50HZ
                           (3): AeAntibandingMode_60HZ - GST_NVCAM_AEANTIBANDING_60HZ
  exposurecompensation: property to adjust exposure compensation
                        flags: readable, writable
                        Float. Range:              -2 -               2 Default:               0
  aelock              : set or unset the auto exposure lock
                        flags: readable, writable
                        Boolean. Default: false
  awblock             : set or unset the auto white balance lock
                        flags: readable, writable
                        Boolean. Default: false
```

These aren't expose through the jetson-utils API, but you can manually add them to the `nvarguscamerasrc` pipeline here:

https://github.com/dusty-nv/jetson-utils/blob/c373f49cf21ad2cae7e4d7da7c41f4fd6473958f/camera/gstCamera.cpp#L144

After you make code changes, re-run `make && sudo make install`

JonPMorgan, 2021-05-06:
Hi Dusty,

Thank you, that's great.

jpmorgan983, 2021-05-06:
Hi Dusty,

Thanks, that's great.

closed 2021-05-06 by jpmorgan983