dusty-nv/jetson-utils #71: Problem in cudaCrop with roi

andrejlevkovitch, 2021-05-26:
For example, i have image with size 100x100 and I want to remove bellow half of the image. So I need crop the image to 100x50 size with top=0, bottom=50, left=0, right=100. So, if we try to use cudaCrop for this task we need specify: input and output buffers, input size and roi. In our case roi will looks like: roi.x = 0, roi.y = 0, roi.z = 100, roi.w = 50. But we have problem in `launchCrop` function with checking the roi:

- here we calculate size of our output buffer by roi, nothing special
https://github.com/dusty-nv/jetson-utils/blob/baf2a527622f28a2b0269a64a58740fd39633c38/cuda/cudaCrop.cu#L56-L57

- but here we check that roi.z and roi.w is less then input size. But it is wrong - roi.z and roi.w can be equal to input width and input height (respectively), because it is not last col/row, but next from last col/row.
https://github.com/dusty-nv/jetson-utils/blob/baf2a527622f28a2b0269a64a58740fd39633c38/cuda/cudaCrop.cu#L69-L70

dusty-nv, 2021-05-26:
Hi @andrejlevkovitch, I think I see what you mean - can you try changing it to this:

```
if( roi.z > inputWidth || roi.w > inputHeight ) 
```

Then recompile with:

``` bash
cd jetson-inference/build
make
sudo make install
```

Does that fix your problem?   If so I can commit the fix to master repo.

andrejlevkovitch, 2021-05-26:
@dusty-nv yes, I know that fixes the issue, so, please, push it to repo

dusty-nv, 2021-05-27:
@andrejlevkovitch just pushed this change in https://github.com/dusty-nv/jetson-inference/commit/19ed62150b3e9499bad2ed6be1960dd38002bb7d