dusty-nv/jetson-utils #74: VideoOutput window object does not close when cross button in title bar is used. Is it possible to use anything else other than a videoSource object to display the resulting frame?

vladtimug, 2021-06-06:
Hello, I am working on an app to run on the jetson nano and I am using a toggle to trigger the inference of a  segnet model to be displayed on a videoOutput window object. When I turn on the inference using the toggle a new window is created where the result is displayed. The problem is that when I turn off the toggle and I press on the cross button of the previously created window the window keeps displaying even tho I get the `[OpenGL] glDisplay -- the window has been closed` message in the terminal. I would like to either solve this issue for the window to close properly or to transfer the output from the segmentation network to another video element which is not a videoSource element. Could you help me figure this out please?

dusty-nv, 2021-06-07:
Hi @vladtimug, are you continuing to call videoOutput.Render() after you close the window?    Does it keep showing new frames, or only the old one (or black)?

vladtimug, 2021-06-07:
@dusty-nv it keeps showing the last frame and it's frozen like that. It will only close after the app itself will close.

dusty-nv, 2021-06-07:
Can you try to 'del output' where output is your videoOutput object?  Or if you are using C++, 'delete output'

vladtimug, 2021-06-26:
Unfortunately that won't get the behavior I'm after because I used a single videoOutput object. Deleting that will stop the video displaying entirely. I only need to kill the window and keep the video stream running.