dusty-nv/jetson-utils #75: How to use jetson-utils as submodule

fduxvtech, 2021-07-02:
Hey, great work with this library and jetson-inference.
I have some problem using this library as submodule, may I have some kind of assistance? I also think that the CMakeList is broken (it uses definition that should not exist) but I'm a novice in using CMake.

Thanks

dusty-nv, 2021-07-02:
Hi @fduxvtech, I would just install jetson-inference, which will also install jetson-utils.  Then you can use jetson-utils from external projects like this:  https://github.com/dusty-nv/jetson-inference/blob/master/examples/my-recognition/CMakeLists.txt

francescodonato, 2021-07-02:
Hi @dusty-nv, I'm from a different account(same person).

I'd l like to modify some stuff and rebuild when needed instead of installing every time, so my question about the submodule if that make sense. I don't know a better way than using the submodule.

dusty-nv, 2021-07-02:
If you add it in your main CMakeLists.txt like here, does it build?   https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/CMakeLists.txt#L164

You may also need to set these like I do in the jetson-inference/CMakeLists.txt

```
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_SOURCE_DIR}/utils/cuda" )
find_package(CUDA)
message("-- CUDA version: ${CUDA_VERSION}")

set(
	CUDA_NVCC_FLAGS
	${CUDA_NVCC_FLAGS}; 
    -O3 
	-gencode arch=compute_53,code=sm_53
	-gencode arch=compute_62,code=sm_62
)

if(CUDA_VERSION_MAJOR GREATER 9)
	message("-- CUDA ${CUDA_VERSION_MAJOR} detected, enabling SM_72")

	set(
		CUDA_NVCC_FLAGS
		${CUDA_NVCC_FLAGS}; 
		-gencode arch=compute_72,code=sm_72
	)
endif()

# setup project output paths
set(PROJECT_OUTPUT_DIR  ${PROJECT_BINARY_DIR}/${CMAKE_SYSTEM_PROCESSOR})
set(PROJECT_INCLUDE_DIR ${PROJECT_OUTPUT_DIR}/include)

file(MAKE_DIRECTORY ${PROJECT_INCLUDE_DIR})
file(MAKE_DIRECTORY ${PROJECT_OUTPUT_DIR}/bin)

message("-- system arch:  ${CMAKE_SYSTEM_PROCESSOR}")
message("-- output path:  ${PROJECT_OUTPUT_DIR}")

set(CMAKE_RUNTIME_OUTPUT_DIRECTORY ${PROJECT_OUTPUT_DIR}/bin)
set(CMAKE_LIBRARY_OUTPUT_DIRECTORY ${PROJECT_OUTPUT_DIR}/lib)
set(CMAKE_ARCHIVE_OUTPUT_DIRECTORY ${PROJECT_OUTPUT_DIR}/lib)

# build C/C++ library
include_directories(${PROJECT_INCLUDE_DIR} ${PROJECT_INCLUDE_DIR}/jetson-inference ${PROJECT_INCLUDE_DIR}/jetson-utils)
include_directories(/usr/include/gstreamer-1.0 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/include /usr/include/glib-2.0 /usr/include/libxml2 /usr/lib/aarch64-linux-gnu/glib-2.0/include/)
```

francescodonato, 2021-07-02:
Ok I'll try this in a few days, I'll return to you eventually, but thanks for now!

closed 2021-07-21 by fduxvtech