dusty-nv/jetson-utils #81: About Cuda Memory

fduxvtech, 2021-07-28:
Hi @dusty-nv,
I was looking into the code of the ring buffer and I saw that in case of zero-copy you are using Mapped Memory, may I ask if using Unified Memory (cudaMallocManaged) will be beneficial?
I did read [this](https://forums.developer.nvidia.com/t/difference-between-cudamallocmanaged-and-zero-copy-memory-function/58660) but I would like to understand specifically about the pro and cons of Unified Memory in the Jetson Xavier hardware.

Thanks!

dusty-nv, 2021-07-28:
On Jetson, there is no PCIe copy with mapped memory, because the CPU and GPU share the same physical memory.  So on Jetson, mapped memory is optimal.

You could also use cudaMallocManaged() to achieve a similar thing - in theory cudaMallocManaged() should know it doesn't need to do memory copies on Jetson.   However that is explicitly the case with mapped memory, since I'm never calling cudaMemcpy().

Also my implementation using mapped memory was around before cudaMallocManaged() came out - for new codebases, I would use cudaMallocManaged() instead for portability.

fduxvtech, 2021-07-28:
I see, I had the same idea but a confirmation is always nice, thank you!

closed 2021-07-28 by fduxvtech