dusty-nv/jetson-utils #83: camera_viewer failed

deeprun, 2021-08-18:
Hi

I got the following message when I run camera_viewer on Jetson Xavier NX:

`[gstreamer] gstreamer message stream-start ==> pipeline0
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:656 Failed to create CaptureSession
[gstreamer] gstCamera -- end of stream (EOS)

(video-viewer:15238): GStreamer-CRITICAL **: 16:42:25.382: gst_mini_object_set_qdata: assertion 'object != NULL' failed
`

The camera is https://www.amazon.com/gp/product/B07T43K7LC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

dusty-nv, 2021-08-18:
@deeprun does your camera work with `nvgstcapture-1.0` program?   If not, camera is plugged in wrong or is some issue outside of this project that you can post to the [Jetson Nano forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76).

deeprun, 2021-08-19:
Below is the error I got when running: nvgstcapture-1.0

e = 4000000
Encoder Profile = High
Encoder control-rate = 1
Encoder EnableTwopassCBR = 0
Opening in BLOCKING MODE
Opening in BLOCKING MODE 
** Message: 18:36:52.366: <main:4670> iterating capture loop ....
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
^C** Message: 18:37:46.215: <_intr_handler:4261> User Interrupted.. 

Terminating the camera pipeline ...

(nvgstcapture-1.0:13734): GStreamer-CRITICAL **: 18:37:47.374: gst_mini_object_set_qdata: assertion 'object != NULL' failed
(Argus) Error Timeout:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 215)
(Argus) Error Timeout: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 102)
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:639 Failed to create CameraProvider
** Message: 18:37:52.388: <main:4680> Capture completed
** Message: 18:37:52.388: <main:4729> Camera application will now exit

deeprun, 2021-08-19:
I plug the camera in the right side of the CSI connector, should I provide any parameter when running nvgstcapture-1.0?

dusty-nv, 2021-08-19:
You shouldn't need any argument to nvgstcapture-1.0

Can you check this article/video to make sure you have the cable plugged in correctly?  https://www.jetsonhacks.com/2019/04/02/jetson-nano-raspberry-pi-camera/

deeprun, 2021-09-09:
@dusty-nv turned out the camera was bad. Replaced it with a new camera and it worked fine. Thank you very much for your help...