dusty-nv/jetson-utils #87: Increasing encoding framerate

ralpioxxcs, 2021-08-24:
Hello!

I tested `video-viewer` following command.
```
./video-viewer \
	--input_URI csi://0 \
	--input_width=1920 \
	--input_height=1080 \
	--input_rate=60 \
	--output_URI file://test_60fps.mp4 \
	--bitrate=10000000 \
	--headless
```
I want to get **60fps**, I changed here to **30** to **60**
https://github.com/dusty-nv/jetson-utils/blob/5c8a0d72f02833b373ce4d779152c4838259b77a/codec/gstEncoder.cpp#L156

also, I remove log message here for decreasing latency
https://github.com/dusty-nv/jetson-utils/blob/5c8a0d72f02833b373ce4d779152c4838259b77a/video/video-viewer/video-viewer.cpp#L117

but, the result is only about 34fps. `gst-discover -v test_60fps.mp4` message here
```
Topology:
  container: video/quicktime
    video: video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)4.2, profile=(string)constrained-baseline, width=(int)1920, height=(int)1080, framerate=(fraction)100000/2787, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)bt709, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
      Tags:
        video codec: H.264 / AVC
        maximum bitrate: 23791680
        bitrate: 9590647
        datetime: 2021-08-24T14:21:38Z
        container format: Quicktime

      Codec:
        video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)4.2, profile=(string)constrained-baseline, width=(int)1920, height=(int)1080, framerate=(fraction)100000/2787, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)bt709, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
      Additional info:
        None
      Stream ID: 251edacab81c11703570e4910ed6113a9ac24e6d8ca6ff1560ebc5f3dc63e3f1/001
      Width: 1920
      Height: 1080
      Depth: 24
      Frame rate: 100000/2787
      Pixel aspect ratio: 1/1
      Interlaced: false
      Bitrate: 9590647
      Max bitrate: 23791680

Properties:
  Duration: 0:00:07.545333333
  Seekable: yes
  Live: no
  Tags:
      video codec: H.264 / AVC
      maximum bitrate: 23791680
      bitrate: 9590647
      datetime: 2021-08-24T14:21:38Z
      container format: Quicktime
```

I think it is possible to loss frame that's because the camera source is a real-time, but the same is true of the video file.

but, it works well at 30fps.

Is it because of the low performance of the Jetson nano ?

Thanks!

## environment
* Jetson nano
* L4T 32.4.4

dusty-nv, 2021-08-24:
It may be due to overhead in gstreamer of using appsrc/appsink with those large HD buffers.  If that's the case I'm not exactly sure how to resolve it other than you using a self-contained gstreamer pipeline (i.e. with gst-launch-1.0)

ralpioxxcs, 2021-08-25:
Thanks for answer @dusty-nv

As you said, I could get 60fps using `gst-launch-1.0`. 
I guess too it's because of the overhead between the appsrc and appsink.. so I changed appsink's callback to action signal to pull buffers at  its own rate.
(i.e `g_signal_emit_by_name (appsink, "pull-sample", &ret)`)

according to [this](https://gstreamer.freedesktop.org/documentation/app/appsink.html?gi-language=c#appsink::pull-sample),

> Note that when the application does not pull samples fast enough, the queued samples could consume a lot of memory, especially when dealing with raw video frames. It's possible to control the behaviour of the queue with the "drop" and "max-buffers" properties.

In my case, it's not a real time application, so I could solve problem.

closed 2021-08-25 by ralpioxxcs