dusty-nv/jetson-utils #94: gst::Encoder::Close() has two seconds of sleep() in it - any way to avoid?

alvervalleysoftware, 2021-09-22:
gstEncoder::Close() has two sleep(1) calls.  In my application, we might be writing multiple video streams simultaneously, and any of them can close at any time.  The sleep()s cause the whole application to 'sleep' for two seconds each time which is causing problems.

I tried removing the sleep()s, but the written video files are then incomplete and unreadable. 

I've modified the sleep()s to be usleep(), and have got them down to 50,000 microseconds, which is an improvement.  But I'm wondering if there is a better way to do this, or to avoid the whole application locking up.  I could use a thread for each write, but is there a simpler way?

Thanks in advance.

alvervalleysoftware, 2021-09-23:
Thanks @dusty-nv - I'll see if I can get that set up.  Thanks for confirming it's necessary, anyway.

ricknitsche, 2022-05-09:
@dusty-nv it seems that the python GIL is locked during these two seconds, which is potentially breaking multithreaded python code using this API. Could this be solved by adding
```
Py_BEGIN_ALLOW_THREADS
// do things
Py_END_ALLOW_THREADS
```
in the python bindings .cpp files to release the GIL? (like [here](https://github.com/rapidsai/custrings/issues/223))

dusty-nv, 2022-05-09:
Hi @ricknitsche, if you have an application that is exhibiting this issue then I would recommend testing that code change, as I don't personally have a repro case for it here.

ricknitsche, 2022-05-10:
> Hi @ricknitsche, if you have an application that is exhibiting this issue then I would recommend testing that code change, as I don't personally have a repro case for it here.

It improves performance (a lot), calling these functions in python threads after this change: https://github.com/dusty-nv/jetson-utils/pull/123