dusty-nv/jetson-utils #102: `jetson.utils.videoSource` Hangs While Trying to Connect to an RTSP Camera

Neltherion, 2021-12-04:
Hi Dusty @dusty-nv 

I'm trying to connect to a Milesight camera via RTSP in Python.

Whenever the `jetson.utils.videoSource(url)` is reached, the following outputs are shown and then the code gets stuck in a blocking mode manner and doesn't move forward from here:

```
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for 192.168.32.128
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 279
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 279 
```

This doesn't happen with my other cameras (a Grundig & CSI Cameras) but happens with this specific camera. 

If I try the `video-viewer` app, the first time I have the above outputs written and the application doesn't work, but if I stop the app with `Ctrl+Z` and then start it again, the application is able to connect to the camera and show it. Unfortunately this can't happen with the Python code since the `jetson.utils.videoSource(url)` part gets stuck...

I've already read problems associated with the `NvMMLiteBlockCreate : Block : BlockType = 279` output but so far the only options are to recompile the latest version of the project or use a variety of different gstreamer plugins that could or could not solve the problem (which seems to also need another recompile of the project).

Is this a known issue and can it be fixed without having to recompile the project?

Thanks as always Dusty

Neltherion, 2021-12-04:
I recompiled/reinstalled the project and the problem is still there...

Neltherion, 2021-12-04:
Here's another related issue: https://github.com/dusty-nv/jetson-inference/issues/615

dusty-nv, 2021-12-06:
Can you try recompiling with `ENABLE_NVMM=off`?  Like this:

``` bash
cd /path/to/your/jetson-inference/build
cmake -DENABLE_NVMM=off ../
make
sudo make install
```

Neltherion, 2022-02-26:
Leaving this comment here so that it may help others: My problem lied in an old Camera Firmware that needed to be updated.