dusty-nv/jetson-utils #106: Stream mp4 file as if it was live (drop frames if necessary)

Kajatin, 2022-01-18:
Hello @dusty-nv ,

When I stream the input from an mp4 file, it looks as if it was processing every frame from the video file, which makes the application look like it was running in slow motion. My pipeline can handle about 10 FPS, but when I stream from a 30 FPS mp4 file, it'll essentially run 1/3 of the speed of _real time_.

Is there a way to configure the gstreamer pipeline string to force it to play it back _live_ and instead skip/drop frames that are not processed/queried in time?

Thanks!
Roland

https://github.com/dusty-nv/jetson-utils/blob/0d3f59f5c0967a108ec1cd58a518ee5ad817d35b/codec/gstDecoder.cpp#L456

dusty-nv, 2022-01-18:
Hi @Kajatin, gstDecoder and the ringbuffer code that it uses actually already does return the latest frame available and skip/drop frames that aren't processed in time.   So my guess is that the entire system is slowing down as opposed to the video reader getting ahead.

What is the resolution of your mp4 file?   You may want to try using a smaller resolution with the[ `--input-width`](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options) and [`--input-height`](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options) arguments.  This can significantly reduce the overhead (and the DNN models take in small resolution anyways)

Kajatin, 2022-01-18:
It's interesting; I haven't thought of the whole system slowing down. It's just that looking at the frames it doesn't seem to be skipping any of them, that's why I thought that it was simply not dropping them when the consumer was slow.

I will investigate whether it's possible that the whole system is just slow.

The video file has a resolution of 2560x1920 @ 20FPS. I have a custom pipeline, so I'm only linking the `jetson-utils` library and use some of the functionality it provides, including setting up an RTSP stream.

johnnynunez, 2024-05-06:
@dusty-nv is it possible to keep all frames with flag? I mean if I want process all frame due benchmarks, datasets etc

dusty-nv, 2024-05-06:
videoSource interface uses a ringbuffer internally to limit memory usage - so just use jetson_utils.cudaMemcpy(img) for each img from capture() and save it as long as you want

johnnynunez, 2024-05-06:
so, @dusty-nv if i doo a loop:
```python3
while True:
    if not self.paused:
        try:
            rgb_img = self.frame_capture.Capture()
            print("\n")
            print(f"Frame: {self.frame_capture.GetFrameCount()}\n")
            # here should I do:
             jetson_utils.cudaMemcpy(img)
 ```
 but it is possible that "while" go to next frame? I mean Capture is decoding on background, not stop if memory is full, dropping frames?

dusty-nv, 2024-05-06:
IIRC it will keep going in background as fast as gstreamer publishes them.  Jetson_utils is meant for live streaming, not seeking/pausing

johnnynunez, 2024-05-06:
> IIRC it will keep going in background as fast as gstreamer publishes them. Jetson_utils is meant for live streaming, not seeking/pausing
> […](#)

thank you dustin!