dusty-nv/jetson-utils #110: Memory leak problem in gstDecoder::Capture

Enigma228322, 2022-01-28:
Hi #dusty-nv!
I've got problem in using your tools related to video capturing. Actually, is video memory keeps cleaning when i use gstDecoder::Capture?
Code:
```cpp
videoOptions opts;
opts.resource = "file:///home/jet/video/1.mp4";
gstDecoder * decoder = gstDecoder::Create(opts);

void * dev_ptr;
imageFormat format = IMAGE_RGB32F;
while (decoder->IsEOS()) {
    decoder->Capture(&dev_ptr,format,0)
}
```
I didnt find any tool to clean allocated memory.
Thank you!

dusty-nv, 2022-01-28:
Hi @Enigma228322, gstDecoder uses a ringbuffer of frames that it re-uses, so it shouldn't be allocating new memory each frame.  This ringbuffer gets free'd when you `delete decoder`

Enigma228322, 2022-01-31:
Thank you @dusty-nv for answer! However, i leaved jeston nano running this code for 60 hours, and its GPU memory grew up from 0.5 GB to 1.9 GB.
But now it settled down and keeps spamming this log:
```
(myjetson:12776): GStreamer-CRITICAL **:09:24:26.228: gst_poll_read_control: assertion 'set != NULL` failed
```

dusty-nv, 2022-01-31:
Hi @Enigma228322, I let a video file run for several hours and was unable to reproduce the memory leak.  This is the command I used:

```
video-viewer --input-loop=-1 /usr/share/visionworks/sources/data/pedestrians.mp4
```