dusty-nv/jetson-utils #111: [question] Does it make sense to add support for video streaming input over UDP?

harleylara, 2022-02-04:
Hello,

I am working with a swarm of Tello EDU drones, which have a front camera that transmits video over UDP. I changed a bit the source code of `videoSource` and `gstDecoder` to support `udp` as input protocol `videoSource('udp://@@:port')`. The idea is to use jetson nano for inference using as input the video stream coming from the drones. Here is the result after modifying the above mentioned code and using directly udp as input protocol [video demo](https://wiki.eolab.de/doku.php?id=drones:mini_drones:start)

My question is, does it make sense to add support for video streaming input over UDP? if the answer is yes, I could open a pull request with the changes.

dusty-nv, 2022-02-04:
Hi @harleylara, is it using RTP with UDP transport?   Can you show me what the GStreamer pipeline looks like?   This pipeline gets printed out after you create the `videoSource` object

harleylara, 2022-02-07:
Hi @dusty-nv, initially I thought the drone was streaming video using RTP with UDP transport as you describe, however when I initialize the `video-viewer` tool like this `video-viewer rtp://<drone-ip>:port` I simply `failed to capture video frame`, but when I use this pipeline using gstreamer `gst-launch-1. 0 -v udpsrc port=<drone video port> ! h264parse ! omxh264dec ! video/x-raw ! autovideosink` I get the streaming video.

The pipeline I get when I run videoSource with the RTP protocol is as follows:

```
udpsrc port=<drone video port> multicast-group=0.0.0.0.0 auto-multicast=true caps="application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name(string)H264" ! rtph264depay ! h264parse ! omxh264dec ! video/x-raw(memory:NVMM) ! appsink name=mysink
```

The problem I observe when using the above pipeline is that gstreamer fails when trying to `depay` the stream because the streaming video is not in RTP ``format```.

The pipeline I get with my modified version of videoSource to support UDP is the following (works perfectly even with video-viwer):

```
udpsrc port=<drone video port> address=0.0.0.0.0 auto-multicast=true ! h264parse ! omxh264dec ! video/x-raw(memory:NVMM) ! appsink name=mysink
```

dusty-nv, 2022-02-07:
That's interesting, I haven't seen networked streaming before with just `udpsrc` and no RTP/RTSP or other protocol on top.

Although I like to support as many protocols as possible, my concern is not having a way to duplicate/test this here to confirm it's working (and with future code updates/ect), so for now I would recommend to just use your patched version of videoSource.   I typically have other machines or cameras setup that act as the RTP/RTSP sources for me so that I can keep testing them with new builds.

harleylara, 2022-02-07:
I have the same concern as you, I'm worried about not being able to have other systems streaming video using only UDP to validate that the code works reliably with other systems, that's why I decided to ask about it before opening a pull request.

Anyway in case someone comes here wanting to use a system that streams video using UDP the proposed solution without changing the source code is to use the following pipeline to take as input UDP and produce a video streaming in RTP, which can be used as input in `videoSource`:

```
gst-launch-1. 0 -v udpsrc port=<UDP input port> ! h264parse ! rtph264pay ! udpsink host=<RTP output ip> port=<RTP output port>
```

It is possible to verify that the above command works correctly using the `video-viwer`:

```
video-viewer rtp://<RTP ip from previous command>:<RTP port from previous command> --input-codec=h264
```

Thanks @dusty-nv for the support.

closed 2022-02-07 by harleylara