dusty-nv/jetson-utils #113: How to reverse the vedio stream

Moresweet, 2022-02-15:
My camera outpouts is inverse when running on my board.I try modfing "gstCamera.cpp" as follow:
`//code in 140
if(mOptions.flipMethod == videoOptions::FLIP_NONE)
    mOptions.flipMethod = videoOptions::FLIP_ROTATE_NONE;// orign is: mOptions.flipMethod = videoOptions::FLIP_ROTATE_180;
`
$ ./imagenet-camera --network=googlenet
After compile finishing, It only has a static image frame. And the console always output "[gstreamer] gstDecoder -- failed to retrieve next image buffer imagenet: failed to capture next frame"
My English is poor and I am a student, so the question I ask may be stupid. I hope getting a solution. Thanks very much.

dusty-nv, 2022-02-15:
Hi @Moresweet, you shouldn't need to manually edit gstCamera.cpp anymore, you can just run the programs with `--input-flip=rotate-180` argument:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options

Moresweet, 2022-02-16:
Thanks for your reply. But it shows same scene above when I use "`--input-flip=rotate-180`". Although the image frame is inverse, it only has one static image and output errors.

dusty-nv, 2022-02-16:
Can you try other --input-flip settings to see if they have any impact?

Can you provide the console log with the errors?  Thanks.

Moresweet, 2022-02-16:
Thank you. I try all the '--input-flip' paramters. Except of `--input-flip=rotate-180`, others all sucessful pass. When I use `--input-flip=rotate-180` the error logs follow: 
```shell
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=0 ! video/x-raw(memory:NVMM) ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0
[0;32m[video]  created gstCamera from csi://0
[0m------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- codec:      raw
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  640x480
[OpenGL] glDisplay -- X window resolution:    640x480
[OpenGL] glDisplay -- display device initialized (640x480)
[0;32m[video]  created glDisplay from display://0
[0m------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      640
  -- height:     480
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   1

[TRT]    TensorRT version 8.0.1
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[... 259 lines cut ...]
[0m[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mimagenet:  failed to capture next frame
[0m[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mimagenet:  failed to capture next frame
[0m[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mimagenet:  failed to capture next frame
[0m[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mimagenet:  failed to capture next frame
[0m[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mimagenet:  failed to capture next frame
[0m[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mimagenet:  failed to capture next frame
[0mreceived SIGINT
[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mimagenet:  failed to capture next frame
[0mimagenet:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
[gstreamer] gstCamera -- pipeline stopped
imagenet:  shutdown complete.
received SIGINT
received SIGINT
```

dusty-nv, 2022-02-16:
Which version of JetPack are you on?   Can you try it with `--input-flip=none`   (that actually equates to rotate-180 because of [this code](https://github.com/dusty-nv/jetson-utils/blob/7de695d94271ba1ae77a4358f427c0887b09084f/camera/gstCamera.cpp#L138)

It should then flip it 180 degrees, because `none` and `rotate-180` get swapped in the code.

Moresweet, 2022-02-17:
My JetPack version is 4.6.  And I has read this source code. My first modification is here too. The `--input-flip=none` is work correctly because it will rotate in the source code.