dusty-nv/jetson-utils #114: Stream to HTTP

qhdwight, 2022-02-25:
Hi, would it be possible to add streaming to HTTP?

dusty-nv, 2022-02-25:
Hi @qhdwight, do you have a working gstreamer pipeline that does this?

Do you mean to make it viewable from a web browser?

qhdwight, 2022-02-27:
Currently do not have a working gstreamer pipeline but looking around I see it is possible.

https://gist.github.com/tetkuz/0c038321d05586841897

Web viewer is ideal! We are planning on putting the stream into an existing UI which is all web tech.

tabiosg, 2022-03-05:
I'll bump... it doesn't looking that bad to add streaming to HTTP for video-viewer based on the above linked forum. This functionality would be useful if added!

dusty-nv, 2022-03-07:
Yea, agreed - this is on my list to look into, and also the web viewer part of it

dusty-nv, 2022-03-08:
Can anyone get https://gist.github.com/tetkuz/0c038321d05586841897 to work?

I'm unable to reproduce it - the browser (Chrome / Chromium) doesn't show the video stream, it just stays blank...