dusty-nv/jetson-utils #116: Build dependency missing

cmoore42, 2022-03-25:
Minor thing, but when checking out the master branch on a fresh Jetson Nano the build of video-viewer fails because libglew is missing.  Maybe you could update the Readme to mention this dependency?  "apt install libglew-dev" fixes the build error.

dusty-nv, 2022-03-25:
Hi @cmoore42, had you installed just jetson-utils?   jetson-inference installs this automatically during the cmake step:  https://github.com/dusty-nv/jetson-inference/blob/01a395892ecc8acdbec4d8e9d6e8ac676416a507/CMakePreBuild.sh#L37

Normally these two libraries are used hand-in-hand, and I don't have an automated dependency installer script just for jetson-utils.  I should document these standalone use-cases better in the readme.

cmoore42, 2022-03-25:
I had installed jetson-inference but was running into errors.  I noticed that jetson-inference was pointing to a particular CL, not to the top of the tree, so I thought maybe there was a bug in jetson-utils that had been fixed and I tried to build it standalone.

I still can't get any of jetson-inference to work with a fresh Jetpack install on a Nano, but I'm not yet sure if there's a bug there , if it's user error on my part, or if I have a hardware issue.  I've been trying to narrow it down.

dusty-nv, 2022-03-25:
Hmm ok - can you let me see the error log and which version of JetPack you are using?   I have tested it on recent versions.  Also there is the jetson-inference docker container you can try.