dusty-nv/jetson-utils #118: gstDecoder -- failed to retrieve next image buffer

masip85, 2022-03-31:
Hello.

I've looking into this issue in forum and github. Normally it happens after an error which is the cause, But,the only error I can see is not retrieving any image. What is happening?

Same issue here. But disabling that option doesn't solve it
```
IN CONTAINER:
root@263a942a2f44:~/repos/jetson-utils/build/aarch64/bin# ./video-viewer rtsp://X:X@192.168.121.181 fileideo.mp4 --headless --input-rtsp-latency=2000URI -- missing/invalid IP port from rtsp://X:X@192.168.121.181, default to port 554
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for X
Opening in BLOCKING MODE
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- discovered video resolution: 3072x1728  (framerate 0.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)5.1, profile=(string)main, pixel-aspect-ratio=(fraction)1/1, width=(int)3072, height=(int)1728, framerate=(fraction)0/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://X:X@192.168.121.181 latency=2000 ! queue ! rtph264depay ! h264parse ! omxh264dec ! video/x-raw(memory:NVMM) ! appsink name=mysink
[video]  created gstDecoder from rtsp://X:X@192.168.121.181
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://X:X@192.168.121.181
     - protocol:  rtsp
     - location:  X
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      h264
  -- width:      3072
  -- height:     1728
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=4000000 ! video/x-h264 !  h264parse ! qtmux ! filesink location=fileideo.mp4
[video]  created gstEncoder from file:///root/repos/jetson-utils/build/aarch64/bin/fileideo.mp4
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: file:///root/repos/jetson-utils/build/aarch64/bin/fileideo.mp4
     - protocol:  file
     - location:  fileideo.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     output
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  30.000000
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from NULL to READY ==> h264parse1
[gstreamer] gstreamer changed state from NULL to READY ==> rtph264depay1
[... 300 lines cut ...]
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager/GstRtpSession:rtpsession0: stats = application/x-rtp-session-stats, rtx-drop-count=(uint)0, sent-nack-count=(uint)0, recv-nack-count=(uint)0, source-stats=(GValueArray)< "application/x-rtp-source-stats\,\ ssrc\=\(uint\)3305988739\,\ internal\=\(boolean\)true\,\ validated\=\(boolean\)true\,\ received-bye\=\(boolean\)false\,\ is-csrc\=\(boolean\)false\,\ is-sender\=\(boolean\)false\,\ seqnum-base\=\(int\)-1\,\ clock-rate\=\(int\)-1\,\ octets-sent\=\(guint64\)0\,\ packets-sent\=\(guint64\)0\,\ octets-received\=\(guint64\)0\,\ packets-received\=\(guint64\)0\,\ bitrate\=\(guint64\)0\,\ packets-lost\=\(int\)0\,\ jitter\=\(uint\)0\,\ sent-pli-count\=\(uint\)0\,\ recv-pli-count\=\(uint\)0\,\ sent-fir-count\=\(uint\)0\,\ recv-fir-count\=\(uint\)0\,\ sent-nack-count\=\(uint\)0\,\ recv-nack-count\=\(uint\)0\,\ have-sr\=\(boolean\)false\,\ sr-ntptime\=\(guint64\)0\,\ sr-rtptime\=\(uint\)0\,\ sr-octet-count\=\(uint\)0\,\ sr-packet-count\=\(uint\)0\;", "application/x-rtp-source-stats\,\ ssrc\=\(uint\)1174459676\,\ internal\=\(boolean\)false\,\ validated\=\(boolean\)true\,\ received-bye\=\(boolean\)false\,\ is-csrc\=\(boolean\)false\,\ is-sender\=\(boolean\)true\,\ seqnum-base\=\(int\)-1\,\ clock-rate\=\(int\)90000\,\ rtp-from\=\(string\)192.168.121.181:15344\,\ rtcp-from\=\(string\)192.168.121.181:15345\,\ octets-sent\=\(guint64\)0\,\ packets-sent\=\(guint64\)0\,\ octets-received\=\(guint64\)19808524\,\ packets-received\=\(guint64\)14699\,\ bitrate\=\(guint64\)11197406\,\ packets-lost\=\(int\)295\,\ jitter\=\(uint\)56\,\ sent-pli-count\=\(uint\)0\,\ recv-pli-count\=\(uint\)0\,\ sent-fir-count\=\(uint\)0\,\ recv-fir-count\=\(uint\)0\,\ sent-nack-count\=\(uint\)0\,\ recv-nack-count\=\(uint\)0\,\ have-sr\=\(boolean\)true\,\ sr-ntptime\=\(guint64\)16568725907500987056\,\ sr-rtptime\=\(uint\)3132578257\,\ sr-octet-count\=\(uint\)14562639\,\ sr-packet-count\=\(uint\)10783\,\ sent-rb\=\(boolean\)true\,\ sent-rb-fractionlost\=\(uint\)5\,\ sent-rb-packetslost\=\(int\)265\,\ sent-rb-exthighestseq\=\(uint\)59299\,\ sent-rb-jitter\=\(uint\)38\,\ sent-rb-lsr\=\(uint\)4033712488\,\ sent-rb-dlsr\=\(uint\)149829\,\ have-rb\=\(boolean\)false\,\ rb-fractionlost\=\(uint\)0\,\ rb-packetslost\=\(int\)0\,\ rb-exthighestseq\=\(uint\)0\,\ rb-jitter\=\(uint\)0\,\ rb-lsr\=\(uint\)0\,\ rb-dlsr\=\(uint\)0\,\ rb-round-trip\=\(uint\)0\;" >, rtx-count=(uint)0;
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager/GstRtpSession:rtpsession0: stats = application/x-rtp-session-stats, rtx-drop-count=(uint)0, sent-nack-count=(uint)0, recv-nack-count=(uint)0, source-stats=(GValueArray)< "application/x-rtp-source-stats\,\ ssrc\=\(uint\)3305988739\,\ internal\=\(boolean\)true\,\ validated\=\(boolean\)true\,\ received-bye\=\(boolean\)false\,\ is-csrc\=\(boolean\)false\,\ is-sender\=\(boolean\)false\,\ seqnum-base\=\(int\)-1\,\ clock-rate\=\(int\)-1\,\ octets-sent\=\(guint64\)0\,\ packets-sent\=\(guint64\)0\,\ octets-received\=\(guint64\)0\,\ packets-received\=\(guint64\)0\,\ bitrate\=\(guint64\)0\,\ packets-lost\=\(int\)0\,\ jitter\=\(uint\)0\,\ sent-pli-count\=\(uint\)0\,\ recv-pli-count\=\(uint\)0\,\ sent-fir-count\=\(uint\)0\,\ recv-fir-count\=\(uint\)0\,\ sent-nack-count\=\(uint\)0\,\ recv-nack-count\=\(uint\)0\,\ have-sr\=\(boolean\)false\,\ sr-ntptime\=\(guint64\)0\,\ sr-rtptime\=\(uint\)0\,\ sr-octet-count\=\(uint\)0\,\ sr-packet-count\=\(uint\)0\;", "application/x-rtp-source-stats\,\ ssrc\=\(uint\)1174459676\,\ internal\=\(boolean\)false\,\ validated\=\(boolean\)true\,\ received-bye\=\(boolean\)false\,\ is-csrc\=\(boolean\)false\,\ is-sender\=\(boolean\)true\,\ seqnum-base\=\(int\)-1\,\ clock-rate\=\(int\)90000\,\ rtp-from\=\(string\)192.168.121.181:15344\,\ rtcp-from\=\(string\)192.168.121.181:15345\,\ octets-sent\=\(guint64\)0\,\ packets-sent\=\(guint64\)0\,\ octets-received\=\(guint64\)19831974\,\ packets-received\=\(guint64\)14717\,\ bitrate\=\(guint64\)11197406\,\ packets-lost\=\(int\)295\,\ jitter\=\(uint\)38\,\ sent-pli-count\=\(uint\)0\,\ recv-pli-count\=\(uint\)0\,\ sent-fir-count\=\(uint\)0\,\ recv-fir-count\=\(uint\)0\,\ sent-nack-count\=\(uint\)0\,\ recv-nack-count\=\(uint\)0\,\ have-sr\=\(boolean\)true\,\ sr-ntptime\=\(guint64\)16568725907500987056\,\ sr-rtptime\=\(uint\)3132578257\,\ sr-octet-count\=\(uint\)14562639\,\ sr-packet-count\=\(uint\)10783\,\ sent-rb\=\(boolean\)true\,\ sent-rb-fractionlost\=\(uint\)5\,\ sent-rb-packetslost\=\(int\)265\,\ sent-rb-exthighestseq\=\(uint\)59299\,\ sent-rb-jitter\=\(uint\)38\,\ sent-rb-lsr\=\(uint\)4033712488\,\ sent-rb-dlsr\=\(uint\)149829\,\ have-rb\=\(boolean\)false\,\ rb-fractionlost\=\(uint\)0\,\ rb-packetslost\=\(int\)0\,\ rb-exthighestseq\=\(uint\)0\,\ rb-jitter\=\(uint\)0\,\ rb-lsr\=\(uint\)0\,\ rb-dlsr\=\(uint\)0\,\ rb-round-trip\=\(uint\)0\;" >, rtx-count=(uint)0;
^Chandling interrupt.
Interrupt: Stopping pipeline ...
EOS on shutdown enabled -- Forcing EOS on the pipeline
Waiting for EOS...
[... cut ...]
(gst-launch-1.0:10288): GStreamer-CRITICAL **: 11:18:29.587: gst_caps_fixate: assertion 'GST_IS_CAPS (caps)' failed
(gst-launch-1.0:10288): GStreamer-CRITICAL **: 11:18:29.587: gst_caps_get_structure: assertion 'GST_IS_CAPS (caps)' failed
(gst-launch-1.0:10288): GStreamer-CRITICAL **: 11:18:29.587: gst_structure_get_string: assertion 'structure != NULL' failed
(gst-launch-1.0:10288): GStreamer-CRITICAL **: 11:18:29.587: gst_mini_object_unref: assertion 'mini_object != NULL' failed*
And what is this?

masip85, 2022-03-31:
Hello
I am using another camera of our collection, and now is getting frames, but bad quality (blurry/huge pixels zone). I've tried default encoding and mpeg4 too. It's also like an slow motion video. (Log doesn't go very very fast). Can't I get better quality?

dusty-nv, 2022-04-01:
I'm not sure why OpenCV says it can't find your appsink element (although I'm not an expert on OpenCV), although this part of your pipeline:

```
! "video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12" ! appsink
```

I think you will want to change to:

```
! "video/x-raw, width=(int)1920, height=(int)1080, format=(string)NV12" ! appsink
```

because I have special code in jetson-utils that handles NVMM memory, but OpenCV does not.

masip85, 2022-04-02:
> Hi @masip85 <https://github.com/masip85>, my guess is some issue or
> connectivity on the RTSP source side. It also appears this RTSP source has
> high resolution of 3072x1728. Maybe you want to try lower resolution at
> first?
>
> If you just run a normal gstreamer pipeline (without the
> hardware-accelerated omx decoder), does it work?

My opencv c++ software works fine but slow with opencv.
My gstreamer pipeline works by itself ( I add a reencoding an mp4 sink.(
bad pixel quality)
My guess is opencv does not admit NV12 ( with any resolution, same result).
Why does video viewer use this? If i want to obtain decoded raw image, is
this the best approach?

Regards

dusty-nv, 2022-04-04:
> Why does video viewer use this?

I use it because the NVIDIA codecs output it and I have a CUDA function that converts NV12 -> RGB.  

However, you can add `nvvidconv` element to your pipeline to convert to the following formats:

```
format: { (string)I420, (string)UYVY, (string)YUY2, (string)YVYU, (string)NV12, (string)NV16, (string)NV24, (string)P010_10LE, (string)GRAY8, (string)BGRx, (string)RGBA, (string)Y42B }
```

(check `gst-inspect-1.0 nvvidconv` for more options)