dusty-nv/jetson-utils #121: H264 decoder for RTSP stream inside the docker

tulseebisen, 2022-05-04:
Hello Everyone! Hope You are doing good…
It’s urgent…I am working on Nvidia Jetson AGX Xavier within Dockerized container…I want to take input from RTSP stream…it’s encoding type is H264 and .avi video input.Input stream frame size is 1920x1080 (in code I am resizing that into 1280x720)

I have used this GStreamer pipeline

cap = cv2.VideoCapture(‘rtspsrc location=“rtsp_link” latency=200 ! queue ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! appsink’, cv2.CAP_GSTREAMER)

It is able to read the frames but after reading frames, frames got completely blurred ,that’s why object detection is not happening.
Kindly help me to solve this issue ASAP. Thank You!

dusty-nv, 2022-05-04:
> cap = cv2.VideoCapture(‘rtspsrc location=“rtsp_link” latency=200 ! queue ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! appsink’, cv2.CAP_GSTREAMER)

Hi @tulseebisen, since your issue is unrelated to jetson-utils library, I suggest that you post this to the Jetson AGX Xavier forum:  https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-agx-xavier/75

tulseebisen, 2022-05-04:
@dusty-nv Oh..Thank You!

closed 2022-05-04 by tulseebisen