dusty-nv/jetson-utils #124: Camera error

spcrobocar, 2022-05-12:
I tried to use the following RPI camera with NX, Aftrer build this repo, when I run camera-viewer, I got the following error. How can I fix it?

aaa@AAA:~/apps/jetson-utils/build/aarch64/bin$ ./camera-viewer
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera – attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw(memory:NVMM) ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0

camera-viewer: successfully initialized camera device (1280x720)
[OpenGL] glDisplay – X screen 0 resolution: 2560x1600
[OpenGL] glDisplay – X window resolution: 2560x1600
[OpenGL] glDisplay – display device initialized (2560x1600)
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
[gstreamer] gstreamer message stream-start ==> pipeline0
camera-viewer: camera open for streaming
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image
^Creceived SIGINT
^Creceived SIGINT
^Creceived SIGINT
[gstreamer] gstDecoder – failed to retrieve next image buffer
camera-viewer: failed to capture RGBA image

camera-viewer: shutting down…
[gstreamer] gstCamera – stopping pipeline, transitioning to GST_STATE_NULL
^Creceived SIGINT

(camera-viewer:10328): GStreamer-CRITICAL **: 22:34:06.816: gst_mini_object_set_qdata: assertion ‘object != NULL’ failed
^Creceived SIGINT
(Argus) Error Timeout: (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 215)
(Argus) Error Timeout: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 102)
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:639 Failed to create CameraProvider
[gstreamer] gstCamera – pipeline stopped
camera-viewer: shutdown complete.
aaa@AAA:~/apps/jetson-utils/build/aarch64/bin$ dmesg | grep -i imx219
[ 2.011313] imx219 9-0010: tegracam sensor driver:imx219_v2.0.6
[ 2.036285] imx219 10-0010: tegracam sensor driver:imx219_v2.0.6
[ 2.628153] tegra194-vi5 15c10000.vi: subdev imx219 9-0010 bound
[ 2.629088] tegra194-vi5 15c10000.vi: subdev imx219 10-0010 bound

dusty-nv, 2022-05-12:
Can you check that the camera is functioning properly first using the `nvgstcapture-1.0` tool?

spcrobocar, 2022-05-12:
Here is the output:

nvidia@XavierNx:~/apps/CSI-Camera$ nvgstcapture-1.0 
Encoder null, cannot set bitrate!
Encoder Profile = High
Supported resolutions in case of ARGUS Camera
  (2) : 640x480
  (3) : 1280x720
  (4) : 1920x1080
  (5) : 2104x1560
  (6) : 2592x1944
  (7) : 2616x1472
  (8) : 3840x2160
  (9) : 3896x2192
  (10): 4208x3120
  (11): 5632x3168
  (12): 5632x4224

Runtime ARGUS Camera Commands:

  Help : 'h'
  Quit : 'q'
  Set Capture Mode:
      mo:<val>
          (1): image
          (2): video
  Get Capture Mode:
      gmo
  Set sensor orientation:
      so:<val>
          (0): none
          (1): Rotate counter-clockwise 90 degrees
          (2): Rotate 180 degrees
          (3): Rotate clockwise 90 degrees
  Get sensor orientation:
      gso
  Set sensor mode:
      smo:<val> e.g., smo:1
  Get sensor mode:
      gsmo
  Set Whitebalance Mode:
      wb:<val>
          (0): off
          (1): auto
          (2): incandescent
          (3): fluorescent
          (4): warm-fluorescent
          (5): daylight
          (6): cloudy-daylight
          (7): twilight
          (8): shade
          (9): manual
  Get Whitebalance Mode:
      gwb
  Set Saturation (0 to 2):
      st:<val> e.g., st:1.25
  Get Saturation:
      gst
  Set Exposure Compensation (-2 to 2):
      ec:<val> e.g., ec:-2
  Get Exposure Compensation:
      gec
  Set Auto Whitebalance Lock:
      awbl:<val> e.g., awbl:0
  Get Auto Whitebalance Lock:
      awbl
  Set Auto Exposure Lock:
      ael:<val> e.g., ael:0
  Get Auto Exposure Lock:
      gael
  Set TNR Mode:
      tnrm:<val> e.g., tnrm:1
          (0): OFF
          (1): FAST
          (2): HIGH QUALITY
  Get TNR Mode:
      gtnrm
  Set TNR Strength (-1 to 1):
      tnrs:<val> e.g., tnrs:0.5
  Get TNR Strength:
      gtnrs
  Set EE Mode:
      eem:<val> e.g., eem:1
          (0): OFF
          (1): FAST
          (2): HIGH QUALITY
  Get EE Mode:
      geem
  Set EE Strength (-1 to 1):
      ees:<val> e.g., ees:0.5
  Get EE Strength:
      gees
  Set Auto Exposure Anti-Banding (0 to 3):
      aeab:<val> e.g., aeab:2
          (0): OFF
          (1): MODE AUTO
          (2): MODE 50HZ
          (3): MODE 60HZ
  Get Auto Exposure Anti-Banding:
      gaeab
  Set Gain Range:
      gr:<val><space><val> e.g., gr:1 16
  Get Gain Range:
      ggr
  Set Exposure Time Range:
      etr:<val><space><val> e.g., etr:34000 35000
  Get Exposure Time Range:
      getr
  Set ISP Digital Gain Range:
      dgr:<val><space><val> e.g., dgr:2 152
  Get ISP Digital Gain Range:
      gdgr
  Capture: enter 'j' OR
           followed by a timer (e.g., jx5000, capture after 5 seconds) OR
           followed by multishot count (e.g., j:6, capture 6 images)
           timer/multihot values are optional, capture defaults to single shot with timer=0s
  Start Recording : enter '1'
  Stop Recording  : enter '0'
  Video snapshot  : enter '2' (While recording video)
  Get Preview Resolution:
      gpcr
  Get Image Capture Resolution:
      gicr
  Get Video Capture Resolution:
      gvcr

Runtime encoder configuration options:

  Set Encoding Bit-rate(in bytes):
      br:<val> e.g., br:4000000
  Get Encoding Bit-rate(in bytes):
      gbr
  Set Encoding Profile(only for H.264):
      ep:<val> e.g., ep:1
          (0): Baseline
          (1): Main
          (2): High
  Get Encoding Profile(only for H.264):
      gep
  Force IDR Frame on video Encoder(only for H.264):
      Enter 'f' 

bitrate = 4000000
Encoder Profile = High
Encoder control-rate = 1
Encoder EnableTwopassCBR = 0
Opening in BLOCKING MODE
Opening in BLOCKING MODE 
** Message: 15:22:34.050: <main:4670> iterating capture loop ....
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 4 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 59.999999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...
ERROR on bus: by /GstPipeline:capture_native_pipeline/GstBin:cap_bin/GstNvArgusCameraSrc:nvarguscamerasrc0: CANCELLED
debug info:
Argus Error Status

(nvgstcapture-1.0:12583): GStreamer-CRITICAL **: 15:22:37.052: gst_mini_object_set_qdata: assertion 'object != NULL' failed
GST_ARGUS: Cleaning up
** Message: 15:22:37.801: <main:4680> Capture completed
** Message: 15:22:37.801: <main:4729> Camera application will now exit

dusty-nv, 2022-05-13:
Does it show the camera?   If not, my code won't work either, and there is an issue with your camera or how it's plugged in.

stelzo, 2022-08-10:
I had the same issue with the log "invalid argument (error 11) (hex 0x0B)" in line 319 gstBufferManager.cpp.
```cpp
if (CUDA_FAILED(cudaGraphicsEGLRegisterImage(&eglResource, eglImage, cudaGraphicsRegisterFlagsReadOnly)))
  return false;
```
I changed the flag from `cudaGraphicsRegisterFlagsReadOnly` to `cudaGraphicsRegisterFlagsNone` and now it works. It was only a guess and I don't know why this flag is invalid but maybe somebody knows or can use this "_fix_".

TX2 -- Jetpack 4.2.2 -- L4T 32.2.1