dusty-nv/jetson-utils #126: Native VPI Support

ligaz, 2022-05-20:
I'm experimenting with the VPI API and was wondering do you plan to support it "natively" without the need to go through the `jetson.utils.cudaToNumpy()` hoops? 

It looks like the [`vpi.asimage(...)`](https://docs.nvidia.com/vpi/python/interoperabilities.html) method is expecting an object with [`__cuda_array_interface__`](https://numba.readthedocs.io/en/stable/cuda/cuda_array_interface.html) so if `PyCudaImage` has one it will work out of the box.

dusty-nv, 2022-05-20:
Hi @ligaz, I haven't looked into this yet but think it's a good idea.  I think a lot of it will depend on if I'm able to implement the numba `__cuda_array_interface__` from C or not.  Maybe it will be enough to just implement the dict as they wish...

ligaz, 2022-12-05:
@dusty-nv It looks like this has been implemented in https://github.com/dusty-nv/jetson-inference/commit/8b499817381cd7ac3f72c7c8742e6a29c1b696c0

closed 2022-12-05 by ligaz