dusty-nv/jetson-utils #132: gstEncoder::Render returns true in case of "pipeline full, skipping frame". Should it return false instead?

slovak194, 2022-06-30:
Hi!
I'm not sure if this is a bug or intended behavior.
My initial expectation was to receive false in case of "full pipeline" to be able to give it another try.

https://github.com/dusty-nv/jetson-utils/blob/56d57f6713a5afe9029ab768ec5d13c285a52dfb/codec/gstEncoder.cpp#L435-L440

Here is what VideoOutput interface says: 

https://github.com/dusty-nv/jetson-utils/blob/56d57f6713a5afe9029ab768ec5d13c285a52dfb/video/videoOutput.h#L166-L168

It is not clear if this is an error or warning situation. If it is still a warning, then an additional method shall be added to check if VideoOutput is ready to receive the next frame. Otherwise, I'd return false in this case.

Best regards,
Alex

JackeyGuo, 2022-09-26:
Hi, can you solve this problem? I encountered this error in x86 with input=video and output=rtp.

slovak194, 2022-09-27:
Hi, @JackeyGuo. I've just changed true to false in the place I mentioned in my initial post. This solved the issue and I can retry pushing to the pipeline later if it is already full.

JackeyGuo, 2022-09-28:
@slovak194 This is not your initial post? can you elaborate on which parameter's TRUE is changed to FALSE.

JackeyGuo, 2022-11-21:
Thank you for your reply.