dusty-nv/jetson-utils #135: Close/Disable rendering window

Unweptpit, 2022-08-15:
Hello, I'm trying to record video to a file. If I start the recording a window opens up showing the camera feed, can I disable that? I do not need it. I'm also using a GUI to record multiple 1 minute clips, every time a new clip is being recorded it opens a new camera feed and doesn't close the old one so if I'm recording 30 one-minute clips then by the end of it I have 30 windows open of old camera feeds which are all frozen. Only when I close the GUI or shutdown the entire python program then the windows close as well. I've tried using `display.Close()` and `del display` but neither do anything. `display` comes from `display = jetson.utils.videoOutput(filename)`

dusty-nv, 2022-08-15:
Hi @Unweptpit, try creating your display object like this instead:

``` python
jetson.utils.videoOutput(filename, argv=['--headless'])
```

Then it shouldn't open the camera window any more.

Unweptpit, 2022-08-15:
Hello, that seems to have worked. Thank you

closed 2022-08-15 by Unweptpit