dusty-nv/jetson-utils #138: how to convert rgb back to nv12?

thunder95, 2022-10-11:
found no direct way to implement this conversion? is it possible to do this indirectly?

for example, convert rgb firstly to yv12 with RGBToYV12, and then convert yv12 to nv12 with the following code?

template <typename T>
__global__ void pack_uv(T * __restrict__ u, T * __restrict__ v, T * __restrict__ uv, const int w, const int h, const int pitch_uv, const int pitch_u, const int pitch_v){

  int idx = threadIdx.x+blockDim.x*blockIdx.x;
  int idy = threadIdx.y+blockDim.y*blockIdx.y;
  if ((idx < w) && (idy < h)){
    T *o  = (T *)(((char *)uv) + idy*pitch_uv);
    T *iu = (T *)(((char *)u)  + idy*pitch_u);
    T *iv = (T *)(((char *)v)  + idy*pitch_v);
    int idx2 = idx >> 1;
    o[idx] = (idx&1)?iv[idx2]:iu[idx2];}
}

dusty-nv, 2022-10-11:
Hi @thunder95, I haven't done RGB -> NV12, but my understanding is that YV12 is quite similar to NV12 - I'm not sure if your function is correct or not, but that would seem to be a good starting point.

thunder95, 2022-10-12:
thanks for your quick reply.

closed 2022-10-12 by thunder95