dusty-nv/jetson-utils #139: Error during building while linking shared library aarch64/lib/libjetson-utils.so

devonsuper, 2022-11-12:
Building fails on linking aarch64/lib/libjetson-utils.so

```
~$ make

[  1%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o
[  2%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o
[  3%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaCrop.cu.o
[  5%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaDraw.cu.o
[  6%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFilterMode.cu.o
[  7%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFont.cu.o
[  9%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaGrayscale.cu.o
[ 10%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaNormalize.cu.o
[ 11%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaOverlay.cu.o
[ 13%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaPointCloud.cu.o
[ 14%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaRGB.cu.o
[ 15%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaResize.cu.o
[ 17%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-affine.cu.o
[ 18%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-fisheye.cu.o
[ 19%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-intrinsic.cu.o
[ 21%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-NV12.cu.o
[ 22%] Building NVCC (Device) object CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YUYV.cu.o
Scanning dependencies of target jetson-utils
[ 23%] Building CXX object CMakeFiles/jetson-utils.dir/URI.cpp.o
[ 25%] Building CXX object CMakeFiles/jetson-utils.dir/XML.cpp.o
[ 26%] Building CXX object CMakeFiles/jetson-utils.dir/commandLine.cpp.o
[ 27%] Building CXX object CMakeFiles/jetson-utils.dir/filesystem.cpp.o
[ 28%] Building CXX object CMakeFiles/jetson-utils.dir/logging.cpp.o
[ 30%] Building CXX object CMakeFiles/jetson-utils.dir/timespec.cpp.o
[ 31%] Building CXX object CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o
[ 32%] Building CXX object CMakeFiles/jetson-utils.dir/camera/v4l2Camera.cpp.o
[ 34%] Building CXX object CMakeFiles/jetson-utils.dir/codec/gstBufferManager.cpp.o
[ 35%] Building CXX object CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o
[ 36%] Building CXX object CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o
[ 38%] Building CXX object CMakeFiles/jetson-utils.dir/codec/gstUtility.cpp.o
[ 39%] Building CXX object CMakeFiles/jetson-utils.dir/cuda/cudaBayer.cpp.o
[ 40%] Building CXX object CMakeFiles/jetson-utils.dir/cuda/cudaColorspace.cpp.o
[ 42%] Building CXX object CMakeFiles/jetson-utils.dir/cuda/cudaPointCloud.cpp.o
[ 43%] Building CXX object CMakeFiles/jetson-utils.dir/display/glBuffer.cpp.o
[ 44%] Building CXX object CMakeFiles/jetson-utils.dir/display/glCamera.cpp.o
[ 46%] Building CXX object CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o
[ 47%] Building CXX object CMakeFiles/jetson-utils.dir/display/glEvents.cpp.o
[ 48%] Building CXX object CMakeFiles/jetson-utils.dir/display/glTexture.cpp.o
[ 50%] Building CXX object CMakeFiles/jetson-utils.dir/display/glWidget.cpp.o
[ 51%] Building CXX object CMakeFiles/jetson-utils.dir/image/imageIO.cpp.o
[ 52%] Building CXX object CMakeFiles/jetson-utils.dir/image/imageLoader.cpp.o
[ 53%] Building CXX object CMakeFiles/jetson-utils.dir/image/imageWriter.cpp.o
[ 55%] Building CXX object CMakeFiles/jetson-utils.dir/input/devInput.cpp.o
[ 56%] Building CXX object CMakeFiles/jetson-utils.dir/input/devJoystick.cpp.o
[ 57%] Building CXX object CMakeFiles/jetson-utils.dir/input/devKeyboard.cpp.o
[ 59%] Building CXX object CMakeFiles/jetson-utils.dir/network/IPv4.cpp.o
[ 60%] Building CXX object CMakeFiles/jetson-utils.dir/network/NetworkAdapter.cpp.o
[ 61%] Building CXX object CMakeFiles/jetson-utils.dir/network/Socket.cpp.o
[ 63%] Building CXX object CMakeFiles/jetson-utils.dir/threads/Process.cpp.o
[ 64%] Building CXX object CMakeFiles/jetson-utils.dir/threads/Thread.cpp.o
[ 65%] Building CXX object CMakeFiles/jetson-utils.dir/video/videoOptions.cpp.o
[ 67%] Building CXX object CMakeFiles/jetson-utils.dir/video/videoOutput.cpp.o
[ 68%] Building CXX object CMakeFiles/jetson-utils.dir/video/videoSource.cpp.o
[ 69%] Linking CXX shared library aarch64/lib/libjetson-utils.so
CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o: In function `gstCamera::discover()':
gstCamera.cpp:(.text+0x13e4): undefined reference to `g_type_check_instance_cast'
CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o: In function `gstCamera::init()':
gstCamera.cpp:(.text+0x1940): undefined reference to `g_error_free'
gstCamera.cpp:(.text+0x1960): undefined reference to `g_type_check_instance_cast'
gstCamera.cpp:(.text+0x1a24): undefined reference to `g_type_check_instance_cast'
gstCamera.cpp:(.text+0x1a50): undefined reference to `g_type_check_instance_cast'
CMakeFiles/jetson-utils.dir/codec/gstBufferManager.cpp.o: In function `gstBufferManager::Enqueue(_GstBuffer*, _GstCaps*)':
gstBufferManager.cpp:(.text+0x824): undefined reference to `g_strcmp0'
CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o: In function `gstDecoder::init()':
gstDecoder.cpp:(.text+0x828): undefined reference to `g_error_free'
[... 75 lines cut ...]
glDisplay.cpp:(.text+0x2ab0): undefined reference to `XResizeWindow'
CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o: In function `glDisplay::SetCursor(unsigned int)':
glDisplay.cpp:(.text+0x2c18): undefined reference to `XCreateFontCursor'
glDisplay.cpp:(.text+0x2cc8): undefined reference to `XDefineCursor'
CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o: In function `glDisplay::ProcessEvents()':
glDisplay.cpp:(.text+0x34f4): undefined reference to `XEventsQueued'
glDisplay.cpp:(.text+0x3518): undefined reference to `XNextEvent'
glDisplay.cpp:(.text+0x3558): undefined reference to `XLookupKeysym'
glDisplay.cpp:(.text+0x35e0): undefined reference to `XLookupString'
glDisplay.cpp:(.text+0x3904): undefined reference to `XGetWindowAttributes'
CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o: In function `glDisplay::onEvent(unsigned short, int, int, void*)':
glDisplay.cpp:(.text+0x40f8): undefined reference to `XKeysymToString'
glDisplay.cpp:(.text+0x4198): undefined reference to `XInternAtom'
glDisplay.cpp:(.text+0x41c4): undefined reference to `XInternAtom'
glDisplay.cpp:(.text+0x41f8): undefined reference to `XSendEvent'
glDisplay.cpp:(.text+0x4224): undefined reference to `XKeysymToString'
collect2: error: ld returned 1 exit status
CMakeFiles/jetson-utils.dir/build.make:1134: recipe for target 'aarch64/lib/libjetson-utils.so' failed
make[2]: *** [aarch64/lib/libjetson-utils.so] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

dusty-nv, 2022-11-12:
Hi @devonsuper, I just tried building jetson-utils standalone, and was able to do so after making commit https://github.com/dusty-nv/jetson-utils/commit/717c6c7dad5980a4d5fb33b277fb995cdf8376e9 (which seems unrelated to your errors).  Which version of JetPack-L4T are you running?

devonsuper, 2022-11-12:
R32.7.2 on the jetson nx xavier

 ```
$ cat /etc/nv_tegra_release

R32 (release), REVISION: 7.2, GCID: 30192233, BOARD: t186ref, EABI: aarch64, DATE: Sun Apr 17 09:53:50 UTC 2022
```

dusty-nv, 2022-11-12:
Hmm strange...I was just able to build this on R32.7