dusty-nv/jetson-utils #140: ImportError: cannot import name 'videoSource' from 'jetson_utils' (/home/nvidia/.local/lib/python3.8/site-packages/jetson_utils/__init__.py)

zhangningboo, 2022-11-22:
Env: Jetson NX 5.0.1 JetPack

clone repo: git clone --recursive https://gitee.com/ningdro/jetson-inference.git
install: cd jetson-inference && mkdir build && cmake .. && make -j$(nproc) && sudo make install && sudo ldconfig
sudo ln -s  /user/lib/python3.8/dist-packages/jetson_utils_python.so /usr/lib/jetson_utils_python.so

run command: $ video-viewer --input-width=640 --input-height=480 --input-codec=mjpeg /dev/video0
`video-viewer` command is ok, but i cannot import videoSource:
```shell
nvidia@ubuntu:~/workspace/jetson-utils/build$ python3
Python 3.8.10 (default, Jun 22 2022, 20:18:18)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>
>>>
>>>
>>> from jetson_utils import videoSource
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'videoSource' from 'jetson_utils' (/home/nvidia/.local/lib/python3.8/site-packages/jetson_utils/__init__.py)
>>>
>>>
>>> from jetson_utils import videoSource, videoOutput, logUsage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'videoSource' from 'jetson_utils' (/home/nvidia/.local/lib/python3.8/site-packages/jetson_utils/__init__.py)
>>> from jetson_utils import videoOutput, logUsage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'videoOutput' from 'jetson_utils' (/home/nvidia/.local/lib/python3.8/site-packages/jetson_utils/__init__.py)
>>> from jetson_utils import logUsage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'logUsage' from 'jetson_utils' (/home/nvidia/.local/lib/python3.8/site-packages/jetson_utils/__init__.py)
>>> exit()
```

dusty-nv, 2022-11-22:
Hi @zhangningboo, normally the jetson-inference/jetson-utils python modules get installed under `/usr/lib/python3.8/dist-packages`, so I'm confused why the errors are showing `/home/nvidia/.local/lib/python3.8/site-packages` in your case.

What does running the command below show on your system?

``` bash
$ ls /usr/lib/python3.8/dist-packages/
cv2/                          jetson/                       jetson_inference_python.so    numpy/                        uff/
graphsurgeon/                 Jetson/                       jetson_utils/                 tensorrt/                     uff-0.6.9.dist-info/
graphsurgeon-0.4.6.dist-info/ jetson_inference/             jetson_utils_python.so        tensorrt-8.4.0.8.dist-info/
```

zhangningboo, 2022-11-23:
hi, I first install jeson_interferce project, and I meat this error: `ImportError: cannot import name 'videoSource' from 'jetson_utils'`. than i download `jetson_utils` project, then rebuild `jetson_utils`.
[image]

dusty-nv, 2022-11-23:
OK thanks, I don't know what this other jetson_utils is but that is unfortunate.

zhangningboo, 2022-11-24:
I found it, ^_^, thanks!
[image]

closed 2022-11-24 by zhangningboo