dusty-nv/jetson-utils #141: how to use webrtc in jetson-utils

JackeyGuo, 2022-11-29:
Thank you very much for your work. I want to use WebRTC. I hope you can give a tutorial to use WebRTC.

dusty-nv, 2022-11-29:
@JackeyGuo it's still WIP, but yes there will be tutorial about it

If you have built/installed the `dev` branch of jetson-inference (or `master` of jetson-utils), you can try running something like this:

``` bash
video-viewer /dev/video0 webrtc://@:49567/my_stream
```

Then navigate your browser to `http://$JETSON_IP:49567`

JackeyGuo, 2022-11-30:
Thanks, but i run the master branch of jetson-utils of the newest version with webrtc output. When i navigate my browser with the seeting port, i can not get the webrtc stream in the browser. The error is like this.
[image]
[image]

By the way, my env is dgpu with gstreamer 1.20.4.1. I doubt that if it is a version issue.

dusty-nv, 2022-11-30:
@JackeyGuo have you installed `libgstreamer-plugins-bad1.0-dev` from apt?

I haven't tested this on x86/dGPU, only Jetson platform (aarch64)

JackeyGuo, 2022-12-05:
Yes, but i install gstreamer from source in dgpu/x86, it must contain this plugin.

poett1, 2023-03-08:
I tried the same on Xavier NX today with JP 5.0.2, jetson-inference@dev branch & jetson-utils@master branch with the following result:
```
...
[webrtc] HTTP 10.10.0.239 GET '/'
[webrtc] /
[webrtc] websocket /my_stream -- new connection opened by 10.10.0.239 (peer_id=0)
[webrtc] new WebRTC peer connecting (10.10.0.239, peer_id=0)

(video-viewer:59099): GLib-GObject-WARNING **: 11:50:03.008: g_object_set_is_valid_property: object class 'GstWebRTCBin' has no property named 'latency'
**
ERROR:/home/user/Dev/jetson-inference-dev/utils/codec/gstEncoder.cpp:866:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Bail out! ERROR:/home/user/Dev/jetson-inference-dev/utils/codec/gstEncoder.cpp:866:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Aborted (core dumped)
```
And Chrome throws `net::ERR_CONNECTION_RESET`. Any hints?