dusty-nv/jetson-utils #142: script or steps to add wifi to sdcard image

AndreV84, 2023-01-14:
@dusty-nv 
May I pick your brain for a moment, please?  Do you know what exactly edits need to be made to nx sdcard image filesystem to add a simple wifi connection that will automatically connect on a first boot?
so far I tried adding wpa_supplicant as in the tutorial below but it doesn't seem loaded by default setup of nx sdcard image https://docs.duckietown.org/daffy/opmanual_duckiebot/out/duckiebot_network.html

dusty-nv, 2023-01-15:
Hey @AndreV84, I don't know how to do it with a file, but personally I connect to wifi with nmcli tool:

``` bash
sudo apt-get install network-manager
sudo nmcli device wifi connect <SSID> password <your password>
```

It's persistent on boot.

AndreV84, 2023-01-15:
@dusty-nv ,
Thank you for the update!
However, the situation requires that first boot from sdcard needs the connection already
 as the devkit is deep into robot without option to connect by any cable; 
only to add sdcard can be easily added to the devkit  then remote connection over ssh need to exist already
 thank you for your suggestion though

closed 2023-01-15 by AndreV84

AndreV84, 2023-01-15:
they somehow implemented it at their sdcard
https://github.com/duckietown/duckietown-shell-commands/blob/daffy/init_sd_card/constants.py#L34-L66
but how to apply it to default nvidia supplied sdcard image is a challenge with no solution as it seems

dusty-nv, 2023-01-15:
@tokk-nv knows about SD card images

AndreV84, 2023-01-15:
thank you for the pointer!  posted to their repo for a feedback https://github.com/NVIDIA-AI-IOT/jetcard/issues/45