dusty-nv/jetson-utils #146: Missing dependencies when building on 35.1.0 container images possible issues with opencv

franciscoQM, 2023-02-09:
Having issues compiling the **latest** jetson-utils ([b4d2fef](https://github.com/dusty-nv/jetson-utils/commit/b4d2fefcc0703995629787a5bd2c3ffcdfc6ee49)) when using the **nvcr.io/nvidia/l4t-pytorch:r35.1.0-pth1.13-py3**  container image. I am trying to build my own custom container using **nvcr.io/nvidia/l4t-pytorch:r35.1.0-pth1.13-py3** as the base image.  As a sanity check I used  **dustynv/jetson-inference:r35.1.0** container image. Both containers have similar issues when trying to build jetson-utils.

In the **dustynv/jetson-inference:r35.1.0** container image I clone the latest version of jetson-utils and try to compile but I get the following errors:

```
In file included from /root/jetson-utils/codec/gstWebRTC.h:26,
                 from /root/jetson-utils/codec/gstWebRTC.cpp:23:
/root/jetson-utils/build/aarch64/include/jetson-utils/WebRTCServer.h:26:10: fatal error: libsoup/soup.h: No such file or directory
   26 | #include <libsoup/soup.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/jetson-utils.dir/build.make:279: CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /root/jetson-utils/codec/gstWebRTC.h:26,
                 from /root/jetson-utils/codec/gstEncoder.cpp:24:
/root/jetson-utils/build/aarch64/include/jetson-utils/WebRTCServer.h:26:10: fatal error: libsoup/soup.h: No such file or directory
   26 | #include <libsoup/soup.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from /root/jetson-utils/codec/gstWebRTC.h:26,
                 from /root/jetson-utils/codec/gstDecoder.cpp:24:
/root/jetson-utils/build/aarch64/include/jetson-utils/WebRTCServer.h:26:10: fatal error: libsoup/soup.h: No such file or directory
   26 | #include <libsoup/soup.h>
```

Which I fixed with: 
`apt-get install libsoup-gnome2.4-dev`

But then I get: 
```
In file included from /root/jetson-utils/codec/gstEncoder.cpp:24:
/root/jetson-utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
   30 | #include <gst/webrtc/webrtc.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/jetson-utils.dir/build.make:4363: CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /root/jetson-utils/codec/gstWebRTC.cpp:23:
/root/jetson-utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
   30 | #include <gst/webrtc/webrtc.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/jetson-utils.dir/build.make:4391: CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o] Error 1
In file included from /root/jetson-utils/codec/gstDecoder.cpp:24:
/root/jetson-utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
   30 | #include <gst/webrtc/webrtc.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/jetson-utils.dir/build.make:4349: CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:205: CMakeFiles/jetson-utils.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

I have tried fixing this by running:
`apt-get install libgstreamer-plugins-bad1.0-dev`

Running this fixes the issue but  the command itself exits with the following error(this error would stop my docker build):

```
Errors were encountered while processing:
 /tmp/apt-dpkg-install-Jy26cS/0-libopencv_4.5.4-8-g3e4c170df4_arm64.deb
 /tmp/apt-dpkg-install-Jy26cS/2-libopencv-dev_4.5.4-8-g3e4c170df4_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Although the command fails, it allows me to move forward in the compiling

when I try to make again I get the following error:
```
/root/jetson-utils/codec/gstWebRTC.cpp:29:10: fatal error: json-glib/json-glib.h: No such file or directory
   29 | #include <json-glib/json-glib.h>
```

Which I try to fix with: 
`apt-get install libjson-glib-dev`

Which results in the following error: 

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libgstreamer-plugins-bad1.0-dev : Depends: libopencv-dev (>= 2.3.0) but it is not going to be installed
 libjson-glib-dev : Depends: gir1.2-json-1.0 (= 1.4.4-2ubuntu2) but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```
I am wondering if this is due to Jetpack 5.0.2 including OpenCV 4.5.4 but the docker image manually installs OpenCV 4.5.0 (with CUDA).
Should be I be using an older version of jetson-utils? Is that the reason jetson-inference locks jetson-utils at [8b6c5ca](https://github.com/dusty-nv/jetson-utils/commit/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c)?
Any guidance is greatly appreciated.

dusty-nv, 2023-02-09:
> Is that the reason jetson-inference locks jetson-utils at [8b6c5ca](https://github.com/dusty-nv/jetson-utils/commit/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c)?

Yes, I would use that commit of jetson-utils that the master branch of jetson-inference is currently on.   The dev branch of jetson-inference uses jetson-utils master and I still need to sanity check the container build for that (although that dockerfile / CMakePreBuild.sh has been updated with some of these dependencies)

franciscoQM, 2023-02-09:
I see, the dev branch of jetson-inference has included many of the missing requirements. Since I am not using the new features I will use the commit of jetson-utils that the master branch of jetson-inference is currently on ([8b6c5ca](https://github.com/dusty-nv/jetson-utils/commit/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c)).

Now I am running into the following issue when running CMake:

```
-- detecting Python 3.7...
-- found Python version:  3.8 (3.8.10)
-- found Python include:  /usr/include/python3.8
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.8.so
-- CMake module path:  /jetson-utils/cuda;/jetson-utils/python/bindings
-- NumPy ver. 1.24.2 found (include: /usr/local/lib/python3.8/dist-packages/numpy/core/include)
-- found NumPy version:  1.24.2
-- found NumPy include:  /usr/local/lib/python3.8/dist-packages/numpy/core/include
CMake Error at cuda/FindCUDA.cmake:1802 (add_library):
  add_library cannot create target "jetson-utils-python-38" because another
  target with the same name already exists.  The existing target is a shared
  library created in source directory "/jetson-utils/python/bindings".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:57 (cuda_add_library)

CMake Error at python/bindings/CMakeLists.txt:59 (target_link_libraries):
  Attempt to add link library "jetson-utils" to target
  "jetson-utils-python-38" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

CMake Error at python/bindings/CMakeLists.txt:62 (target_link_libraries):
  Attempt to add link library "npymath" to target "jetson-utils-python-38"
  which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.
```

I am not sure what the problem is now, I saw a similar issue in the [nvidia developer forums](https://forums.developer.nvidia.com/t/fail-to-build-jetson-inference-inside-nvcr-io-nvidia-l4t-ml-r35-1-0-py3/225981/9?u=franciscon9k63) but that didn't help.

The Dockerfile I am running is very similar to that of the Dockerfile in jetson-inference master:
```
FROM nvcr.io/nvidia/l4t-pytorch:r35.1.0-pth1.13-py3

ENV DEBIAN_FRONTEND=noninteractive
ENV SHELL /bin/bash

RUN apt-get update

# install development packages and gstreamer
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
    cmake \
    nano \
    lsb-release \
    gstreamer1.0-tools \
    gstreamer1.0-libav \
    gstreamer1.0-rtsp \
    gstreamer1.0-plugins-rtp \
    gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad \
    gstreamer1.0-plugins-ugly \
    && rm -rf /var/lib/apt/lists/*

# pip dependencies for pytorch-ssd
RUN pip3 install --upgrade pip
RUN pip3 install --upgrade Cython && \
    pip3 install boto3 pandas tensorboard

RUN apt-get update
RUN apt-get install -y dialog
RUN apt-get install -y libglew-dev glew-utils libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev
RUN apt-get install -y qtbase5-dev

# install OpenCV (with CUDA)
ARG OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz
ARG OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz

COPY scripts/opencv_install.sh /tmp/opencv_install.sh
RUN cd /tmp && ./opencv_install.sh ${OPENCV_URL} ${OPENCV_DEB}

# Install jetson-utils requirements

# RUN TZ=Etc/UTC   \ 
#     apt-get update && \
#     apt-get install -y libsoup-gnome2.4-dev \  ***add
#     libgstrtspserver-1.0-dev  \
#     libglew-dev \
#     libjson-glib-dev *** add

##################################################
## CHECKPOINT: Requirements to run jetson-utils ##
##################################################

# Install jetson-utils-qm

RUN git clone https://github.com/dusty-nv/jetson-utils.git && \
    cd jetson-utils &&   \
    git checkout 8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c && \
    mkdir build && cd build &&  \
    cmake .. &&   \
    make -j$(nproc) &&   \
    make install
```

dusty-nv, 2023-02-10:
> -- detecting Python 3.7...

Hi @franciscoQM, it shouldn't attempt to find Python 3.7 on JetPack 5.x because of this check:

https://github.com/dusty-nv/jetson-utils/blob/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c/python/CMakeLists.txt#L8

However it appears that when jetson-utils gets built standalone, `LSB_RELEASE_CODENAME` is never set.   

Please try adding [this](https://github.com/dusty-nv/jetson-inference/blob/1bc949db39c4bc471a185e8c5ec81264ba10e472/CMakeLists.txt#L39) code block to [jetson-utils/CMakeLists.txt:50](https://github.com/dusty-nv/jetson-utils/blob/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c/CMakeLists.txt#L50) so that it gets set in the standalone build.

franciscoQM, 2023-02-11:
Thanks that was the problem. Instead of running CMake like:
```
 cmake ..
```
I ran:
```
 cmake -DLSB_RELEASE_CODENAME="focal" ..
```

closed 2023-02-11 by franciscoQM