dusty-nv/jetson-utils #147: Corrupted frames when reading from file (mp4) on x86

slovak194, 2023-03-07:
Hi!

I'm trying to use this nice repo both on a jetson and on a laptop with an x86 CPU and Nvidia GPU.
However, the very first try gives me broken/corrupted frames even in the simplest example with mp4.
Here is what I do:

``` C++

#include <iostream>

#include "jetson-utils/videoOutput.h"
#include "jetson-utils/videoSource.h"

#include "jetson-utils/cudaMappedMemory.h"

int main() {

  uchar3 *frame = nullptr;
  uchar3 *frame_resized = nullptr;

  videoOptions disp_options;
  disp_options.zeroCopy = false;
//  auto disp = videoOutput::Create("display://0", disp_options);
  auto disp = videoOutput::Create("file:///home/user/out.mp4", disp_options);

//  std::string file_path = "/home/nvidia/Downloads/DellePR1.4.mp4";
  std::string file_path = "/home/user/Downloads/DellePR1.4.mp4";

  std::string jutil_path = "file://" + file_path;

  videoOptions src_options;
  src_options.zeroCopy = false;
  auto cap = videoSource::Create(jutil_path.c_str(), src_options);
  auto new_options = cap->GetOptions();
  new_options.Print();

  uint32_t width = cap->GetWidth();
  uint32_t height = cap->GetHeight();

  if (!cudaAllocMapped((void **) &frame, width, height, imageFormat::IMAGE_RGB8)) {
    fprintf(stderr, "[POOL] -- cudaAllocMapped failed\n");
    exit(1);
  }

  bool res = true;

  while (res) {

    cudaDeviceSynchronize();
    res = cap->Capture(&frame);
    cudaDeviceSynchronize();

    if (res) {
      auto res_render = disp->Render(frame, width, height);
      while (!res_render) {
        std::cout << "retry render\n";
        res_render = disp->Render(frame_resized, width, height);
      }
    }
  }

  disp->Close();

  return 0;
}
```

This snippet works nicely on a jetson, but on x86 I got this:
[image]
Apparently, old memory is used. I've tried to use cudaDeviceSynchronize() but that does not help. I've also used a queue, and an artificial pause and the mapped memory is disabled but the result is the same all the time.

Best regards,
Alex

dusty-nv, 2023-03-07:
Hi @slovak194, as a first test to debug, does this work for you?

``` bash
video-viewer /home/user/Downloads/DellePR1.4.mp4 /home/user/out.mp4
```

What GPU card to you have in your x86 system?   Are you running the jetson-inference x86 container?

Note that x86 support in this repo is pretty much a beta feature (and container only), as I don't personally have the resources to test jetson-inference across the multitude of system/OS configurations that exist for x86.

slovak194, 2023-03-08:
Hi, @dusty-nv 

Thank you for getting back!

> does this work for you?

No, same issue. Frames are corrupted both in out.mp4 file and in a viewer.

> What GPU card to you have in your x86 system?

NVIDIA GeForce 940MX
Driver Version: 530.30.02
Driver Version: 530.30.02    CUDA Version: 12.1
CUDA Toolkit 11.8

> Are you running the jetson-inference x86 container?
I use bare jetson-utils installation from the latest master without inference and containers.

> Note that x86 support in this repo is pretty much a beta feature (and container only), as I don't personally have the resources to test jetson-inference across the multitude of system/OS configurations that exist for x86.

We can do (already doing) that.
I run very common Ubuntu LTS 20.04 and the environment is pretty much the same as on our jetsons. But of course, even the smallest hint from your side will be very useful. I suspect the issue is in how memory is used somewhere in the buffer manager.

We would love to have it working as it gives the possibility to easily move load/algorithms between jetson and PC.
Thank you in advance!

Best regards,
Alex