dusty-nv/jetson-utils #148: The RTSP URL cannot be reconnected after disconnection.

Pi-Boss, 2023-03-29:
I try to enter the command
```
./video-viewer --input-decoder=v4l2 --input-loop=-1 rtsp://admin:123456@192.168.3.22:554/h264/ch1/main/av_stream rtsp://@:8566/test --no-display
```
“I disconnected the camera from the network and then reconnected it, but the app cannot establish a connection to RTSP and keeps printing `[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer` in a loop.To reconnect to RTSP, I have to restart the app after exiting,the option ‘–input-loop’ seems to be invalid for RTSP
I want to automatically retry connecting to RTSP after disconnection.Can you tell me how to achieve this?
I’m looking forward to your reply, this feature is very important to me @dusty-nv

Pi-Boss, 2023-03-29:
this is part of the log

[gstreamer] gstEncoder -- pipeline full, skipping frame 5800 (1920x1080, 3110400 bytes)
video-viewer:  captured 5800 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 5825 (1920x1080, 3110400 bytes)
video-viewer:  captured 5825 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 5850 (1920x1080, 3110400 bytes)
video-viewer:  captured 5850 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 5875 (1920x1080, 3110400 bytes)
video-viewer:  captured 5875 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 5900 (1920x1080, 3110400 bytes)
video-viewer:  captured 5900 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 5925 (1920x1080, 3110400 bytes)
video-viewer:  captured 5925 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 5950 (1920x1080, 3110400 bytes)
video-viewer:  captured 5950 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 5975 (1920x1080, 3110400 bytes)
video-viewer:  captured 5975 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 6000 (1920x1080, 3110400 bytes)
video-viewer:  captured 6000 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 6025 (1920x1080, 3110400 bytes)
video-viewer:  captured 6025 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 6050 (1920x1080, 3110400 bytes)
video-viewer:  captured 6050 frames (1920x1080)
[gstreamer] gstEncoder -- pipeline full, skipping frame 6075 (1920x1080, 3110400 bytes)
video-viewer:  captured 6075 frames (1920x1080)
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder -- end of stream (EOS)
[gstreamer] gstDecoder -- seeking stream to beginning (loop 2 of -1)
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[... cut ...]
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstDecoder::Capture() -- a timeout occurred waiting for the next image buffer

dusty-nv, 2023-03-29:
@Pi-Boss I think after a certain number of timeouts are detected, the easiest way would be to just recreate the videoSource interface.   I recently added a recent `status` argument to videoSource::Capture() which allows you to determine when a timeout occurred:

https://github.com/dusty-nv/jetson-utils/blob/363148fb3558ae58c1b9b21ee2446de2de3355ae/video/videoSource.h#L240

(the Python videoSource.Capture() will return `None` on a timeout)

Pi-Boss, 2023-03-31:
I found that frequently calling videoSource::Create(inputOpt) for a non-existent camera can lead to memory leaks. Although it returns nullptr, the 'dec' object is not released. In addition, the 'discover()' function in 'dec->init()' can also cause memory leaks. Of course, all of these issues only occur when the camera's RTSP connection cannot be established. @dusty-nv

Pi-Boss, 2023-03-31:
"If I simultaneously input 16 channels of RTSP video streams (H264 1920x1080) into Jetson NX and convert them into RTSP Server (H265 1920x1080), will it occupy all my GPU resources? I found that this will severely affect the model inference process of TensorRT inside the device."
[image]

dusty-nv, 2023-03-31:
@Pi-Boss for that many simultaneous streams, you should look at DeepStream - it has higher performance

I will make a note to look into the memory leaks, thanks.

Pi-Boss, 2023-03-31:
You're welcome, the jetson-utils open-source library is very useful, and I look forward to your continuous updates. @dusty-nv