dusty-nv/jetson-utils #149: cudaOverlay() Transparent Pixels

alex-stefan, 2023-03-30:
Hi,

Is it possible to modify the cudaOverlay() so that image files with transparent backgrounds are overlaid correctly?

alex-stefan, 2023-04-04:
Thanks, @dusty-nv ! Your tip on image format allowed me to figure out what to search for and was able to get good results. Took a bit to get my head around the documentation, but it worked well in the end.

From your experience, is it feasible to expect Jetson to be able to process one 1080@60 CSI stream, 1xH264 1080@30 USB camera, overlay a few images on it and then output the results to an SD card at 60FPS?

dusty-nv, 2023-04-04:
> From your experience, is it feasible to expect Jetson to be able to process one 1080@60 CSI stream, 1xH264 1080@30 USB camera, overlay a few images on it and then output the results to an SD card at 60FPS?

Hi @alex-stefan, glad that you got it working - regarding your question about the stream bandwidth, which Jetson are you using, and what kind of processing are you doing (i.e. DNN inference, computer vision, or just displaying it).   Regardless, it sounds doable.

alexxs88, 2023-04-04:
I'm using a Nano. No AI/computer vision for now, just overlaying and displaying it or saving to SD.