dusty-nv/jetson-utils #150: Does video-viewer read the compressed MJPEG image or raw YUYV image when using v4l2 devices?

therishidesai, 2023-04-06:
It looks like when I pass `/dev/video0` to video-viewer I am getting close to 20FPS with my USB camera and looking at the logs it seems like video-viewer is reading the MJPEG stream from v4l2. Is this correct and is there a way to have video-viewer read the raw YUYV stream from v4l2?

dusty-nv, 2023-04-07:
Hi @therishidesai, you can try `--input-codec=raw`.  I would be interested to see the console log from when it starts up and prints out/selects the available camera formats and to know which camera you are using?  Thanks.

therishidesai, 2023-04-07:
Yeah so it looks like video-viewer by default was using the MJPEG. When I used `--input-codec=raw` it was a much lower frame rate which makes sense with what `v4l2-ctl --list-formats-ext` shows.

closed 2023-04-07 by therishidesai