dusty-nv/jetson-utils #151: cannot run cuda-to-pytorch.py

Yiming0416, 2023-04-07:
Hi,

Thanks for your great work! I am trying to run the `cuda-to-pytorch.py`. The line of `print(cuda_img)` works fine but it got stuck at the line of `print(cuda_img.__cuda_array_interface__)` and finally output `Segmentation fault (core dumped)`. Any idea about this?

dusty-nv, 2023-04-08:
Hmm...havent seen this...have you built/installed the master branch of jetson-inference/jetson-utils?

Yiming0416, 2023-04-08:
I am using Jetson nano 4GB with jetpack 4.6.1 and PyTorch 1.10. I built jetson-inference from source following the instruction [here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md). And I am able to run the example code for object detection etc. on the Jetson nano. Do I need to install/build jetson-utils seperately?

dusty-nv, 2023-05-03:
@Yiming0416 this was fixed in https://github.com/dusty-nv/jetson-utils/commit/0bcb19b498326eb866a80d7d13388b2e59bc9dfd (not sure why it only showed up on JetPack 4 and not JetPack 5)

closed 2023-05-03 by dusty-nv