dusty-nv/jetson-utils #155: ImportError: cannot import name 'cudaAllocMapped'

roblockeho, 2023-04-16:
Hey Dustin

I've installed the jetson-utils and when I run the examples I get this:-

`Traceback (most recent call last):
  File "cuda-examples.py", line 26, in <module>
    from jetson_utils import (cudaAllocMapped, cudaConvertColor, cudaCrop,
ImportError: cannot import name 'cudaAllocMapped'`

when I installed, I did cmake, make, sudo make install and sudo ldconfig without errors. 

This is on a TX2, and I compiled with OpenCV 4.4.0 with Cuda enabled.

Have I missed something?

Thanks!

PS, great work, quality videos and articles - very helpful - thank you!

roblockeho, 2023-04-17:
@dusty-nv thanks for your reply, yes I can run this without errors.

```
rob@ubuntu:~/jetson-utils/python/examples$ python3 -c 'import jetson_utils'
rob@ubuntu:~/jetson-utils/python/examples$ python3 cuda-examples.py 
Traceback (most recent call last):
  File "cuda-examples.py", line 26, in <module>
    from jetson_utils import (cudaAllocMapped, cudaConvertColor, cudaCrop,
ImportError: cannot import name 'cudaAllocMapped'
```

dusty-nv, 2023-04-17:
Hmm that's strange, I'm not sure why you can `import jetson_utils` but not find the functions within it.   Can you run any of these?

``` bash
python3 -c 'from jetson_utils import cudaAllocMapped'
python3 -c 'from jetson_utils import cudaConvertColor'
```

If you run the jetson-inference container, do you have the same problems there?   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md

roblockeho, 2023-04-17:
No, sorry...

```
python3 -c 'from jetson_utils import cudaAllocMapped'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: cannot import name 'cudaAllocMapped'
```

```
python3 -c 'from jetson_utils import cudaConvertColor'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: cannot import name 'cudaConvertColor'
```

It's a relatively fresh install of jetpack, I could wipe and reinstall?

dusty-nv, 2023-04-17:
> Considering this is just a warning, I would assume it's imported.

Yes, strange, but seems to be imported.  I would change references of `jetson_utils` to `jetson.utils` as needed while I investigate this further...

roblockeho, 2023-04-17:
I can confirm it's working with jetson.utils.

Thank you for your support, really appreciated!