dusty-nv/jetson-utils #156: Use NNVM memory when the codec is CODE_RAW

panda9095, 2023-04-23:
Hi,

Enabling NNVM memory did not bring performance gain when the codec is CODE_RAW. 

Specifically, the pipeline is as follows: `v4l2src device=/dev/video0   ! video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080 ! appsink name=mysink`.  With that pipeline, `gst_caps_features_contains(gstCapsFeatures, GST_CAPS_FEATURE_MEMORY_NVMM))` returns `false`, then  the CPU copy from `gstBuffer` is used, rather than NNVM memory.

I am wondering if we can use NNVM memory when the codec is CODE_RAW?
Thx

panda9095, 2023-04-25:
@dusty-nv Understood. 
I am using jetson-utils for V4L2 camera capturing (1080p@30fps) on a Jetson Xavier, which works in a MAXN mode. Typically the memcpy cost only 10+ms to copy 6MB data from `gstBuffer`, but the cost time can go to 45ms in some cases. Is there any way to improve the copy peformance? 

Thx.

dusty-nv, 2023-04-25:
Hmm, how frequently does that occur?   Since you are already in MAX-N, have you tried running `jetson_clocks` too?

panda9095, 2023-04-25:
@dusty-nv Yes, the `jetson_clocks` is on too. 
I have two Xaviers. The xavier in office shows stable copy performance (about 10+ms), but the other xavier which is on a AGV reports the slow-copy issue. Here is some log:
`[DEBUG] [1682327771.625429631]: gst2buffer cpy cost 14
[DEBUG] [1682327771.657340061]: gst2buffer cpy cost 13
[DEBUG] [1682327771.690871921]: gst2buffer cpy cost 13
[DEBUG] [1682327771.724910560]: gst2buffer cpy cost 14
[DEBUG] [1682327771.757805779]: gst2buffer cpy cost 14
[DEBUG] [1682327771.792562600]: gst2buffer cpy cost 15
[DEBUG] [1682327771.870543572]: gst2buffer cpy cost 54
[DEBUG] [1682327771.900762552]: gst2buffer cpy cost 18
[DEBUG] [1682327771.919667236]: gst2buffer cpy cost 17
[DEBUG] [1682327771.939682410]: gst2buffer cpy cost 18
[DEBUG] [1682327771.960783371]: gst2buffer cpy cost 16
[DEBUG] [1682327771.994610735]: gst2buffer cpy cost 16
[DEBUG] [1682327772.024462719]: gst2buffer cpy cost 14
[DEBUG] [1682327772.068030729]: gst2buffer cpy cost 21
[DEBUG] [1682327772.092950387]: gst2buffer cpy cost 15
[DEBUG] [1682327772.124845073]: gst2buffer cpy cost 14
[DEBUG] [1682327772.174974454]: gst2buffer cpy cost 30
[DEBUG] [1682327772.193428458]: gst2buffer cpy cost 14
[DEBUG] [1682327772.226540136]: gst2buffer cpy cost 15
[DEBUG] [1682327772.260683036]: gst2buffer cpy cost 16
[DEBUG] [1682327772.292891466]: gst2buffer cpy cost 15
[DEBUG] [1682327772.324445366]: gst2buffer cpy cost 14
[DEBUG] [1682327772.367985181]: gst2buffer cpy cost 24
[DEBUG] [1682327772.391454395]: gst2buffer cpy cost 14
[DEBUG] [1682327772.425777401]: gst2buffer cpy cost 15
[DEBUG] [1682327772.460853952]: gst2buffer cpy cost 16
[DEBUG] [1682327772.495665176]: gst2buffer cpy cost 17
[DEBUG] [1682327772.528703026]: gst2buffer cpy cost 16
[DEBUG] [1682327772.579597664]: gst2buffer cpy cost 35
[DEBUG] [1682327772.604195994]: gst2buffer cpy cost 23
[DEBUG] [1682327772.631831989]: gst2buffer cpy cost 21
[DEBUG] [1682327772.682291244]: gst2buffer cpy cost 19
[DEBUG] [1682327772.709569652]: gst2buffer cpy cost 16
[DEBUG] [1682327772.736462536]: gst2buffer cpy cost 18
[DEBUG] [1682327772.775740909]: gst2buffer cpy cost 31
[DEBUG] [1682327772.802227499]: gst2buffer cpy cost 20
[DEBUG] [1682327772.826399534]: gst2buffer cpy cost 15
[DEBUG] [1682327772.859011890]: gst2buffer cpy cost 14
[DEBUG] [1682327772.893431605]: gst2buffer cpy cost 15
[DEBUG] [1682327772.926534610]: gst2buffer cpy cost 15
[DEBUG] [1682327772.959032208]: gst2buffer cpy cost 15
[DEBUG] [1682327772.991639507]: gst2buffer cpy cost 14
[DEBUG] [1682327773.053437700]: gst2buffer cpy cost 36
[DEBUG] [1682327773.101971540]: gst2buffer cpy cost 45
[DEBUG] [1682327773.143257060]: gst2buffer cpy cost 38
[DEBUG] [1682327773.194106096]: gst2buffer cpy cost 38
[DEBUG] [1682327773.216031708]: gst2buffer cpy cost 17
[DEBUG] [1682327773.238165427]: gst2buffer cpy cost 19
[DEBUG] [1682327773.260354733]: gst2buffer cpy cost 21
[DEBUG] [1682327773.286129829]: gst2buffer cpy cost 17
[DEBUG] [1682327773.306565762]: gst2buffer cpy cost 19
[DEBUG] [1682327773.334529295]: gst2buffer cpy cost 20
[DEBUG] [1682327773.359945620]: gst2buffer cpy cost 16
[DEBUG] [1682327773.394564418]: gst2buffer cpy cost 17
[DEBUG] [1682327773.427492609]: gst2buffer cpy cost 17
[DEBUG] [1682327773.460830334]: gst2buffer cpy cost 16
[DEBUG] [1682327773.494362884]: gst2buffer cpy cost 17
[DEBUG] [1682327773.528675441]: gst2buffer cpy cost 16
[DEBUG] [1682327773.561738431]: gst2buffer cpy cost 17
[DEBUG] [1682327773.591849627]: gst2buffer cpy cost 14
[DEBUG] [1682327773.625788597]: gst2buffer cpy cost 15
[DEBUG] [1682327773.659019724]: gst2buffer cpy cost 15
[DEBUG] [1682327773.693013897]: gst2buffer cpy cost 15
[DEBUG] [1682327773.726197821]: gst2buffer cpy cost 15
[DEBUG] [1682327773.762367303]: gst2buffer cpy cost 16
[DEBUG] [1682327773.792656715]: gst2buffer cpy cost 15
[DEBUG] [1682327773.827684284]: gst2buffer cpy cost 17
[DEBUG] [1682327773.860415066]: gst2buffer cpy cost 16
[DEBUG] [1682327773.892839400]: gst2buffer cpy cost 15
[DEBUG] [1682327773.928279725]: gst2buffer cpy cost 17
[DEBUG] [1682327773.962264074]: gst2buffer cpy cost 17`

dusty-nv, 2023-04-25:
> but the other xavier which is on a AGV reports the slow-copy issue

I'm not sure if the Xavier on your AGV is inside some enclosure or housing, but I wonder how the temperatures are and if some degree of thermal throttling is occurring?    You might want to try keeping an eye on the clocks / system temps with tegrastats or jtop.