dusty-nv/jetson-utils #158: Font Size

KalitheBirb, 2023-05-06:
If you try to Set the Font size to large you will get this error.

[cuda]   cudaOverlayText( mFontMapGPU, maxGlyphSize, mFontMapWidth, ((GlyphCommand*)mCommandGPU) + mCmdIndex, numCommands, color, image, image, format, width, height)
[cuda]      invalid configuration argument (error 9) (hex 0x09)
[cuda]      /home/ai/jetson-inference/utils/cuda/cudaFont.cu:563
[cuda]   cudaOverlayText( mFontMapGPU, maxGlyphSize, mFontMapWidth, ((GlyphCommand*)mCommandGPU) + mCmdIndex, numCommands, color, image, image, format, width, height)
[cuda]      invalid configuration argument (error 9) (hex 0x09)
[cuda]      /home/ai/jetson-inference/utils/cuda/cudaFont.cu:563

I tried the tweak the values in the constructor but was wondering if there is another simple value that I need to change or something to look at.

Sorry if this is something simple still a bit new.

dusty-nv, 2023-05-06:
@KalitheBirb which Jetson are you using?   Due to the way my CUDA kernels are written for the font drawing, I think there's an upper limit on the font size based on how many CUDA threads per block the particular GPU supports.

KalitheBirb, 2023-05-06:
> @KalitheBirb which Jetson are you using? Due to the way my CUDA kernels are written for the font drawing, I think there's an upper limit on the font size based on how many CUDA threads per block the particular GPU supports.

Got a couple, The 4gb Jetson Nano, Jetson Xavier NX, And just got the Jetson Nano Orin. Currently testing on the Orin (Loving the new board Btw!)

KalitheBirb, 2023-05-06:
Project is based on your image classification example and with the new power of the orin been testing 4k input and output. So far working well!

dusty-nv, 2023-05-06:
Ok great!  Glad you are enjoying your new Orin boards :)

Joannou1, 2024-06-11:
Seconding this. I'm trying to overlay text on a 4K sized image and the largest font (45) is just simply too small :(